Sapd/HeadsetControl #1: Can't compile on Linux Mint

zestyr, 2016-02-15:
Great job on this tool, I've been looking for something like this for months, thanks for the effort!

I cloned the repository, install libusb-1.0, but still wasn't able to compile. 
This is the error message I get:

```
Programs/G930SideTone [master] » make
gcc main.c -L"/usr/lib" -lusb-1.0 -o G930Sidetone
main.c: In function ‘find_logitech_g930’:
main.c:110:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (ssize_t i = 0; i < cnt; i++)
     ^
main.c:110:5: note: use option -std=c99 or -std=gnu99 to compile your code
main.c: In function ‘send_sidetone’:
main.c:134:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int i = 16; i < 64; i++)
     ^
make: *** [default] Error 1
```

zestyr, 2016-02-15:
I fixed it by changing this line in the makefile:
`gcc main.c -L"/usr/lib" -lusb-1.0 -o G930Sidetone`
To this:
`gcc main.c -L"/usr/lib" -lusb-1.0 -std=gnu99 -o G930Sidetone`

Sapd, 2016-02-15:
Seems like your compiler is using the old standard as default, I updated the makefile.

I also pushed a udev rule, so that it's possible to access the port without root permissions. Just type sudo make install to install it into the system.

https://github.com/Sapd/G930SideTone/commit/c5a62535457f119d897839347a7b7d625cd03e78

zestyr, 2016-02-15:
Sounds good. 

I know this isn't what your software is set out to do, but would it at all be possible to change the "Bass" and "Treble" values that are exposed in the Logitech Gaming Software? If so, I would very much appreciate it if you could add that feature.

closed 2016-02-15 by zestyr

Sapd, 2016-02-15:
I think the software just applies an equalizer before it goes out to the headset. I don't think that the software sends the values to the headset and the headset applies it.

You can do exactly the same with pulseaudio: https://wiki.archlinux.org/index.php/PulseAudio#Equalizer
Just don't forget to set the equalizer to the default output in sound settings of the system.

zestyr, 2016-02-15:
Yeah that's what I thought, nevermind then. 

I've been using pulseaudio-equalizer for a while, but it doesn't sound nearly as good as in Windows, which is quite a shame. I've played around with a bunch of different settings and the equalizer, but it's just not that good unfortunately.

Sapd, 2016-02-15:
Yes, I also noticed this. But I don't know how the software does it. In theory the same effect should be possible with pulseaudio (even when the stick does this "transformation"). But I guess the driver does some further mixing to make the sound more vibrant.

Maybe the driver uses a hrtf (5 channels emulation) even in stereo modus to give the music more "room".
You can try out hrtf filters: https://www.reddit.com/r/linux_gaming/comments/2ot5ov/enable_system_wide_hrtf_with_pulseaudio/
One advantage is that you then can get also 5.1 virtual sound to work.

My windows 10 driver doesn't seem to recognize the dolby switch any more, so I can't see if there are packets send when using that switch :( (this would show if the stick does this, or the driver).