Sapd/HeadsetControl #2: [Feature Request] Support for Plantronics DA45

dkadioglu, 2016-02-16:
Hi,

I am not sure, if it is really possible but, could you help me in doing the following:
1. Find out, which data has to be sent via USB to set the sidetone of the DA45 (if it supports it at all).
2. Extend your tool to also support the DA45.

I would be very happy about your support.

Thanks in advance.

Sapd, 2016-02-16:
Yes, I planned support for other headsets anyway, and already implemented locally the corsair void headset.

You need a usb sniffer which is capable of reading the buffers in memory (I tried wireshark, but it couldn't). USBlyzer was capable of: http://www.usblyzer.com/

Start the software, and check the port of your headset. Then go the driver software, and if you just started it, wait a bit so all packages you don't need are exchanged.

Then try to set Sidetone to e.g. 100. And look for the packages, the packages will look like their sent multiple times, it's because it sniffs on multiple drivers. Look for outgoing packages which have some raw data in the raw data column. Try to set Sidetone to something lower, and again to 100, to see if it's the same raw data when you put it to 100.
If it is, post me the raw data of Sidetone set to 100 and Sidetone set to 0. I also need the data in the URB tab: RequestType, Request, Value, Index.

It should look like this: https://gist.github.com/Sapd/128b9fbba612467485c5

dkadioglu, 2016-02-16:
Thanks for your reply!

If I didn't make any mistake, here is the requested data:

Sidetone Mute

RawData         01
RequestType     00h
Request         01h
Value           0100h
Index           0700h

Sidetone Unmute

RawData         00
RequestType     00h
Request         01h
Value           0100h
Index           0700h

Sidetone 0

RawData         00 E2
RequestType     00h
Request         01h
Value           0200h
Index           0700h

Sidetone 100

RawData         00 00
RequestType     00h
Request         01h
Value           0200h
Index           0700h

If you need anything else, please ask.

dkadioglu, 2016-02-16:
I just saw, that it is possible to configure the sidetone in ALSA already. So it is maybe not necessary to implement support for that. Sorry for the noise and thank you very much for your support.

Sapd, 2016-02-16:
No problem, but its interesting that alsa is including the functionality. Can you control it via pulseaudio?

dkadioglu, 2016-02-16:
Actually not, I've got to use alsamixer for that. Here is a screenshot:
[image]

closed 2017-06-24 by Sapd