Sapd/HeadsetControl #5: G633 support

Moosieboy, 2016-10-12:
Any chance you could implement support for the Logitech G633 Artemis Spectrum?

Sapd, 2016-10-14:
Yes it's possible, but I need packet data, see https://github.com/Sapd/HeadsetControl/issues/5

Sapd, 2016-10-14:
Ah, I forgot something. It could be, that the packets are the same as it's likely that logitech doesn't change them for every headset.

Type lsusb in the linux command line, and note the vendor and product id of the headset. Then swap out the G930's ones here https://github.com/Sapd/HeadsetControl/blob/master/main.c#L34
recompile and try if you can set the sidetone.

dvponomarev, 2017-11-08:
Hello, 
I tried this with my G633,
set
```
#define VENDOR_LOGITECH 0x046d
#define PRODUCT_G930    0x0a5c
```
in main.c

when I execute sudo headsetcontrol 0 I got:
```
$ sudo headsetcontrol 0
Headsetcontrol written by Sapd (Denis Arnst)
	https://github.com/Sapd
Found Logitech G930!
Error in transfering data :(
```

Could you help please?

Sapd, 2017-11-08:
Likely the protocol is different after all. I need packet data to help you with implementing the G633

ashkitten, 2018-01-08:
@Sapd I have a G933 and was starting to work on reverse engineering the protocol, but I've got no idea where to start. I've committed the .pcapng files to [a repository](https://github.com/ashkitten/g933-utils) so you can take a look if you've got the time. The relevant device is numbered 1.2.

Sapd, 2018-01-10:
Hello,

did you put the Sidetone down and Up with the original driver? 

If yes, packet number 587 in your startup.pcapng could be such a packet. But I am not 100% sure, some of the values:

> 11:ff:04:0d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> 
> 11:ff:04:3d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> 
> 11:ff:04:ed:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> 
> 11:ff:04:3d:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> 
> 11:ff:04:2d:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00

When you look closely, one values seems to always change (the 4th one). 0d, 3d, ed.
So I would try to send packets like the first three lines of my quote. Look closely to the URB setup section of those packets in Wireshark, and use the same parameters when you send those packets over libUSB.

ashkitten, 2018-01-10:
I've figured out that the protocol used for the G933 is at least very similar to HID++ 2.0 as detailed here: https://lekensteyn.nl/files/logitech/

The first byte describes the length of the packet, 0x11 for long (20 bytes)
The second byte describes the destination (0xff for the receiver or default device)
The third byte is the feature index
The fourth byte is a combination of the function ID and the application ID (first nibble and second nibble respectively)
The remaining bytes are arguments or return values

ashkitten, 2018-01-10:
I've added some more packet captures to my repository with the latest commit, those might be very useful

Sapd, 2018-01-25:
Did you made further progress? My suggestion is, that you simply try to send some packets, which you may think could represent the sidetone by trial and error (like the ones I suggested). You could also try analysing with USBlyzer as I find it a little bit more clearer.

ashkitten, 2018-01-25:
I've reverse engineered most of the relevant features, and implemented a library to address those features. There are also my notes, which contain detailed information about everything I've learned so far.

Sapd, 2018-01-25:
I see, it looks good. When I have time again I will rewrite a part of the software to support multiple features of multiple devices (the current architecture with the hard-coded function to the sidetone functions isn't well suited), as I already reverse engineered some more features of the corsair void.

Moosieboy, 2018-01-29:
@Sapd With the information you have on @ashkitten's repo, is it possible for you to implement support for the G933? If not I'll install Windows on my machine and try to get the values.

Sapd, 2018-01-29:
You'll have to test it. You could try using his software and simply modifying the device id's. If it works, it uses the same protocol.

You can btw. also sniff inside an virtual machine (with usb passthrough).

ashkitten, 2018-01-29:
The software I wrote will only work on Linux at the moment, due to using hidraw exclusively. Also, I'd just like to take a moment to mention that I use she/her pronouns, @Sapd :)

Sapd, 2018-03-14:
I've got now time and reworked the complete software. It now also uses hidraw. Hidraw is a better alternative as it in fact works in MacOS when reading values.

Moosieboy, 2018-05-09:
@Sapd I've captured the data necessary for the sidetone for the G633.:

Sidetone 0:
00000000  11 FF 07 1A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sidetone 100:
00000000  11 FF 07 1A 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

RequestType 22
Request 09
Value 0211
Index 0003

I tried adding it manually: https://gist.github.com/Moosieboy/2d1ddfbc9407aa0eb6543a94b617e4d4
But it's saying Failed to set sidetone. Error: -1: (null). I couldn't find where to change the requesttype, request, value & index in the source? Many thanks.

Moosieboy, 2018-05-09:
Yep, it worked when switching to hid_write. I had to leave the first byte in though, otherwise it still failed. Many thanks!

So the final file:
`#include "../device.h"

#include <string.h>

static struct device device_g633;

static int g633_send_sidetone(hid_device *device_handle, uint8_t num);

void g633_init(struct device** device)
{
    device_g633.idVendor = VENDOR_LOGITECH;
    device_g633.idProduct = 0x0a5c;
    
    strcpy(device_g633.device_name, "Logitech G633 Gaming Headset");
    
    device_g633.capabilities = CAP_SIDETONE;
    device_g633.send_sidetone = &g633_send_sidetone;
    
    *device = &device_g633;
}

static int g633_send_sidetone(hid_device *device_handle, uint8_t num)
{
    unsigned char data[5] = {0x11, 0xFF, 0x07, 0x1A, num};
    
    return hid_write(device_handle, data, 5);
}`

Sapd, 2018-05-09:
Great to hear! If you want you can make a pull request. Simply fork the repo on GitHub, make changes (and care a bit for similar code style in regards to the rest of the project), push to your repo and then under Pull Requests you can open one.

Moosieboy, 2018-05-09:
Done. See #18

Moosieboy, 2018-05-09:
Sorry, see #19

closed 2018-05-09 by Moosieboy