Sapd/HeadsetControl #6: Void Wired Support

SethBarberee, 2017-06-20:
I made a  [fork](https://github.com/SethBarberee/HeadsetControl/tree/VoidWired) that implements partial support for the Wired Void. It finds the Void Wired however, the sidetone command for the Void fails with:
```
Found Corsair Wired!
Error in transfering data :(
```
This would lead me to believe that possibly the protocol for the Wired version is different than the Wireless? Any ideas?

Sapd, 2017-06-24:
Sorry, for the late reply.
I didn't correctly write a check to determine the error in the code.
Replace line 201 (in your fork):
`printf("Error in transfering data :(: %d: %s\n", size, libusb_error_name(size));`
Then run again to determine the exact error.

SethBarberee, 2017-06-24:
This was outputted along with the regular error message
`
-9: LIBUSB_ERROR_PIPE
`

Sapd, 2017-06-24:
Yes, this definitely means a packet type is incorrect.
In a Windows System (VM with USB throughput works too) capture the packets and while capturing, put the slidetone slider up to 100% and down to 0% multiple times.
(Install Wireshark and make sure to check USBPcapCMD while installing, go to C:\Program Files\USBPcap start USBPcapCMD.exe, choose the number where your device is and type a filename. Stop with CTRL+C, then upload the file for me).

SethBarberee, 2017-06-24:
https://drive.google.com/open?id=0B-NU0La4w6BDakF5bUh1eVJPUE0

Sapd, 2017-06-24:
Didn't think that the protocol would be completely different.
I tried implementing it and pushed it in a new branch, clone it with `git clone -b void-wired https://github.com/Sapd/HeadsetControl.git` and report back wether it works.

SethBarberee, 2017-06-24:
Error Output:
```
Found Corsair VOID wired!
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=2
Error in transferring data :( -1: LIBUSB_ERROR_IO
```

Sapd, 2017-06-24:
Can you post the output of `libusb -vvv` regarding your device?

SethBarberee, 2017-06-24:
`lsusb -v` from my Arch Linux install:
https://gist.github.com/SethBarberee/0cafd54e7e7db89525a7ec33c2f36712

Sapd, 2017-06-24:
That's weird, your libusb doesn't show the endpoints I see on the wireshark log.

But I also saw other packets, it's basically a wild guess, but try to replace the send_sidetone_void_wired function with this:
```   
    unsigned short loudness = map(num, 0, 128, 0, 65535);

    unsigned char data[2] = {loudness & 0xff, (loudness >> 8) & 0xff};

    int size = libusb_control_transfer(device_handle, LIBUSB_DT_HID, LIBUSB_REQUEST_CLEAR_FEATURE, 0x0200, 2816, data, 2, 2000);

    if (size > 0)
    {
        printf("Set Sidetone successfully! (%d bytes transferred)\n", size);
    }
    else
    {
        printf("Error in transfering data :(: %d: %s\n", size, libusb_error_name(size));
    }
```

SethBarberee, 2017-06-24:
```
Found Corsair VOID wired!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
Error in transfering data :(: -1: LIBUSB_ERROR_IO
```

Sapd, 2017-06-25:
I'm kind of clueless, I transfer it exactly as in the wireshark logs. It's like I'm looking at the wrong device.

You could try to redo the capture and upload it again. After playing around with the slider do for 10 seconds or so nothing before stopping the capture, so I can filter out the packets which don't belong to the sidetone setting. Also don't have any music or so playing while capturing.

SethBarberee, 2017-06-25:
I redid the capture and waited about 10 seconds as requested.
https://drive.google.com/open?id=0B-NU0La4w6BDRXUyZEZNV050QVU

Sapd, 2017-06-25:
Somehow the capture doesn't match the lsusb output. On the capture there are the endpoints 0x82 for input and 0x03 for output. On the lsusb output, there is only 0x84 however. It's like that these are different devices. That's also the reason you get the error, he simply can't find the endpoint. If you already run lsusb and the compiled program as root (with the old transfer interrupt code), I sadly don't know where to look next.

Sapd, 2017-06-25:
I think I've found the error. I booted in a VM into linux and checked my lsusb output, it is actually quite the same as yours. However your headset still uses interrupts.
Then I looked at directions, and noticed that I mismatched OUT and IN.
So use the code of the first change I made (with the interrupt). And replace line 221 with:
`int ret = libusb_interrupt_transfer(device_handle, 0x83, data, 64, &transferred, 1000);`
If this doesn't work, also try out 0x84:
`int ret = libusb_interrupt_transfer(device_handle, 0x84, data, 64, &transferred, 1000);`

Does this (hopefully) work?

SethBarberee, 2017-06-27:
0x83 gave the same IO error:
```
Error in transferring data :( -1: LIBUSB_ERROR_IO
```
**However** 0x84 gave a different error:
```
Error in transferring data :( -7: LIBUSB_ERROR_TIMEOUT
```
I guess that's progress?

Sapd, 2017-06-27:
Could be, but not necessarily. You can try to increase the last number from 1000 to 3000, it's the timeout. And you could also try endpoints 0x81 and 0x82. 

I also checked the packets of my own headset, I noticed that sometimes wireshark shows not every packet, usblyzer instead made it quit easy to find it out.

SethBarberee, 2017-06-27:
0x81 and 0x82 both give the IO error as above.

Increasing the timeout to 3000 gave the same timeout error. I even tried 5000 as the timeout to no avail.

orblazer, 2019-04-27:
Hello,
I try find solution for configure my Corsair Void pro.

You can find dumped data from iCUE and my `lsbus -v` here : https://pastebin.com/NinS2MhW

Thanks.

Sapd, 2019-04-27:
Try to open one of the existing implementations of the Corsair void, and replace the productId with the id of your product (0x0a1e), then recompile

closed 2019-08-12 by Sapd