Sapd/HeadsetControl #8: Logitech G430 on Linux

joeslazaro, 2017-09-27:
I tried again to get this working on Linux, and I discovered additional information dmesg output when I ran headsetcontrol:
```
[523380.230450] usb 3-9: usbfs: process 6312 (headsetcontrol) did not claim interface 0 before use
[523380.232022] input: Logitech Logitech G430 Gaming Headset as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9:1.3/0003:046D:0A4D.0010/input/input27
[523380.290414] hid-generic 0003:046D:0A4D.0010: input,hidraw1: USB HID v1.00 Device [Logitech Logitech G430 Gaming Headset] on usb-0000:00:14.0-9/input3
```

If I change the calls to libusb_claim_interface() and libusb_release_interface() to use 0 for the second argument, then it works.

I don't want to submit a PR with that change because I assume that it would break everything that was already working. Do you think there is a way to dynamically discover the correct value to use depending on device and platform?

joeslazaro, 2017-09-27:
I looked further at the lsusb output, and found that the interface number that I needed (0) corresponds to the entry where bInterfaceClass = 1 Audio && bInterfaceSubClass = 1 Control Device
```
joe@ironforge:~/Projects/HeadsetControl-g430$ lsusb -d 046d: -v

Bus 003 Device 008: ID 046d:0a4d Logitech, Inc. G430 Surround Sound Gaming Headset
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        16
  idVendor           0x046d Logitech, Inc.
  idProduct          0x0a4d G430 Surround Sound Gaming Headset
  bcdDevice            1.20
  iManufacturer           1 Logitech
  iProduct                2 Logitech G430 Gaming Headset
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          253
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
...
```

Sapd, 2017-10-04:
That explains why it works on Mac.
On Mac claiming a device is not needed and not even possible when it's already claimed (which it usually is).

I think when I looked into some other device, it also had a different id.

Instead of using a define, we could make a new struct with information for a device:

```
// vendor is needed multiple times, so we can define it here
#define VENDOR_LOGITECH 0x046d

struct device {
    uint16_t idVendor;
    uint16_t idProduct;

    int interfaceNumber;
}

// and then initialize every "device" with static initialization
struct device device_g930 = { .idVendor = VENDOR_LOGITECH, .idProduct = 0x0a1f, .interfaceNumber = 3};
```

On another commit, we could even insert a function pointer into the struct, so we don't need the if switches in `send_sidetone`.

Such a struct could be also useful later, for having something like a bit mask with driver capabilities (e.g. I implemented locally battery information for corsair void)

Sapd, 2019-12-09:
See #27

closed 2019-12-09 by Sapd