Sapd/HeadsetControl #13: VOID PRO battery check issue

kuchenstrudel, 2018-03-27:
When the mic is put up vertically, which triggers an automatic mute, the battery readings are wrong.

Examples: 

> mic down, muted (with the mute key)
> Battery: 94%

> mic down, unmuted
> Battery: 94%

> mic up, automuted
> Battery: 222%

Maybe this is an issue with the VOID PRO and you are not getting it with yours, if so: can I provide you with any readings to figure this out?

Sapd, 2018-03-27:
Could be a simple flag/bitmask, because 222-94=128. You could try it again with a different battery level, check the level, put the mic up and check the level again and subtract 128 to see if the result is the same.

kuchenstrudel, 2018-03-27:
Unplugged the headset, and spammed `HeadsetControl -b`

(mic up)
> Battery: 222%
 
> Failed to request battery. Error: -100: (null)

> Battery: 223%

> Battery: 223%
 
> Battery: 223%

(mic down)
> Battery: 95%

223-95 -> 128, will leave it unplugged for a while and report back once again.

kuchenstrudel, 2018-03-27:
(mic up)
> Battery: 220%

(mic down)
> Battery: 92%

220-92 -> 128, seems you are correct.

Sapd, 2018-03-27:
Then you can do something like this:
```c
// on top of the c file
enum voidpro_battery_flags {
     VOIDPRO_BATTERY_MICUP = 128
}

// then inside the function
if (data_read[4] & VOIDPRO_BATTERY_MICUP)
    return data_read[4]-VOIDPRO_BATTERY_MICUP;
```

This will simply return the battery status. We could also pass it, however then the battery_status enum has to be updated.

kuchenstrudel, 2018-03-27:
I adjusted your code a bit and it looks good (as in working) now. See https://github.com/Sapd/HeadsetControl/pull/14

kuchenstrudel, 2018-03-28:
Fixed in https://github.com/Sapd/HeadsetControl/commit/5f49bd63d257af57e348760e00e2fe1b21077d7a, closing.

closed 2018-03-28 by kuchenstrudel