Sapd/HeadsetControl #15: G533 macOS (High Sierra 10.13.4)

codmpm, 2018-04-16:
Thank you for this!! I've already asked the Logitech support thousand times for OSX support for the Gaming-Headsets. As I'm also using the headset unser OSX too, besides gaming on a Win10 machine.

Now for the problem: I'm using a G533 and can't set the sidetone like it should be: 

`./HeadsetControl -s 128` disables sidetone completely, which is ok for me.
`./HeadsetControl -s 127` and `./HeadsetControl -s 1` practicaly do the same, can't tell any difference.

I've cloned the current `master` and compiled with make, cmake, etc. via brew.

The problem ist not, that I'm missing the sidetone but that I hear a loud swooshy noise (Rauschen) obviously coming from the sidetone. Under Windows I can get around this by setting the sidetone to ~50%. 

So, currently `-s 128` is working for me without any sidetone at all on OSX.
Is this is a bug or did I understand something wrong?

Cheers and thank you,
Patrik

Sapd, 2018-04-16:
No, normally 128 should be the highest level and 0 no sidetone.

The G533 supported was created by simply adapting the Corsair VOID solution, as it was the same protocol.

I looked into the history and It seems like I omitted a mapping of the values, when I adapted to a new software structure.
Can you try go into _src/devices_ open _logitech_g533.c_ and insert before line 25 (directly after the start of the function) `num = map(num, 0, 128, 200, 255);` and on top of the file `#include "../utility.h"`?

codmpm, 2018-04-16:
Worked like a charm, thank you. 

But there seems something wrong with the info given, if I use `-s 129`: 

    Usage: ./HeadsetControl -s 0-100

Sapd, 2018-04-16:
Thank you, I updated Master.

closed 2018-04-16 by Sapd

codmpm, 2018-04-16:
Especially thank you!

commit 1d30ffce (2018-04-16): Fixed sidetone value-mapping of G533. Resolves #15