Sapd/HeadsetControl #17: Check battery level for g930 headset

gurken-salat, 2018-04-27:
Hi, I followed the guide in #3 to check the protocol for the g930 headset and I hope I did it right.

> Search for packets with a info of URB_INTERRUPT in, if you found some, check the leftover capture data field, this is the data field. Now you need to go through multiple of such packets and see if the values make sense as battery indicator (e.g. if there is a 32 in hex, that could mean that the battery is 50% full). Post me the full data of such a packet (the data and the headers!).

As I cannot find the described pattern I attached the [log file](https://github.com/Sapd/HeadsetControl/files/1954220/g930log.pcap.log) of USBPcap.

During capture, I did the following (time and packet column might be incorrect, I only looked for the right interval which seemed promising)

event, time, leftover capture data, packet no
charge,     101,   02 20 04 00 00,   2820
15s later
discharge,  116,   02 00 04 00 00,   2929
15s later
charge,     131,   02 20 04 00 00,   3044
15s later
discharge,  146,   02 00 04 00 00,   3157
15s later
charge,     161,   02 20 04 00 00,   3287
15s later
discharge,  176,   02 00 04 00 00,   3400
15s later
charge,     191,   02 20 04 00 00,   3530
15s later
discharge,  206,   02 00 04 00 00,   3639

15s later
unplug,     221, ,  3754
5s later
plug
25s later
unplug,     251, ,  4783
5s later
plug

(headset was almost full during capture)

The protocol of g930 seems totally different to me and I cannot find the battery level...

Is there anything else I can provide to have the battery check implemented for the g930?

Sapd, 2018-04-27:
When you look closely (probably you already noticed). The second byte changes when it's plugged in. The Corsair Void also has this information in the same packet, so I would try discharging it and see whether the packet data changes further.

However I would also look at other packet types, when I wrote the guide I didn't try it by myself, I only guessed. In the Corsair Void, the software sends a request for getting battery information and then does a read. In the original software, it happened when the corsair software was opened (e.g. on start or after minimizing).

Sapd, 2019-12-09:
I close the issue as I think the headset is no longer being produced. (But feel free to request to open it again or make a new one).

codydg, 2020-04-15:
I'd be interested to see battery support for the G930.

I've installed this before:
https://github.com/Roliga/g930-battery-percentage
and it worked fine for me. But I haven't yet looked into the source yet. I may try to implement it in the next week or so and submit a pull request.

Sapd, 2020-04-15:
I looked into it, it is quite simple to implement.

When you look into his code, inside main.c, it has one hid_send_feature_report, this is most likely the request for battery status.

He then uses hid_get_feature_report to read the battery. As this function has a timeout, he probably does it multiple times. You probably make a new buffer, but make sure to set the first byte before calling to 0xff (reportid).
Instead you could also directly try hid_read. It will block until one packet is read. However you would have to check in an _if_, wether it is the correct packet.

tobiaspc, 2020-04-21:
I ported that feature and submitted a pull request: #77

closed 2020-05-09 by Sapd