Sapd/HeadsetControl #20: Sidetone stops working after muting mic from headset itself.

alete89, 2018-06-04:
Hi, thanks for bringing this awesome tool. Gaming with friends without sidetone makes me speak too loud, lol.

I have a G533 and HeadsetControl simply stops working after I lift up the mic (this automatically mutes it) when I put back the mic in its position.

Sapd, 2018-06-05:
Hello,
that is a real weird behavior. It would mean, that the Windows software waits for an interrupt, when the mic is put down again and re-sets it's sidetone.

Fixing it, would mean further packet logging is needed. However maybe a simple trick works. You could try setting the sidetone while the mic is up and see whether something weird happens.

If not, you could try setting it every second, like this:
`while sleep 1; do HeadsetControl -s 100; done` 
(Interrupt with CTRL+C)

alete89, 2018-06-06:
`while sleep 1; do HeadsetControl -s 100; done` this makes sidetone work for "a millisecond" and then go away every second. The same if i do `sleep 5` it works for an instant ant then stops for 5 seconds.

> However maybe a simple trick works. You could try setting the sidetone while the mic is up and see whether something weird happens.

I didn't get this, sorry. Would you explain that again for a newbie?

alete89, 2018-06-07:
I just discovered that sidetone also goes away and doesn't turn back after headphones suspend (after some minutes without listening to anything). You can wake them up with the volume scroll, but sidetone doesn't come back.

alete89, 2018-06-13:
Mi linux driver-development skills are null, but I like the challenge, I'm logging the devices ISR right now, hoping that I can identify the one that would trigger the sidetone on again. If I make it, I would love to integrate it with your work.

alete89, 2018-06-14:
It doesn't seem like headset is sending any data when you mute or unmute from either lifting and pulling the mic or pressing the button. I can't intercept any packet when I raise or low volume with the wheel. Maybe I'm doing something wrong. Any advice?

Sapd, 2018-06-14:
> I can't intercept any packet when I raise or low volume with the wheel.

These packets are sent over the audio interface, so you won't see anything if you don't listen to it. (However looking into this interface also doesn't make sense)

> It doesn't seem like headset is sending any data when you mute or unmute from either lifting and pulling the mic or pressing the button.

That is however weird, maybe false interface?

Make sure, that you capture while you are in Windows and the Headset Software is open.

alete89, 2018-06-16:
> Make sure, that you capture while you are in Windows and the Headset Software is open.

I arrived to this conclusion after some hours (lol at me). I couldn't get this done through a VM, anyway. So I guess I'll try again in a Windows host. Thanks!

closed 2019-08-12 by Sapd