Sapd/HeadsetControl #21: Sidetone Not Working For G430 on Mac

VeniVidiViciVersace, 2018-06-18:
After building and installing, I type in the command to disable sidetone (headsetcontrol -s 0), and I receive the following in terminal:
"Found Logitech G430"
"Error: This headset doesn't support sidetone"

It's very possible I built the necessary parts incorrectly, as I'm new to Mac OS. If there is any other information I could provide to facilitate the process, please let me know. Thanks in advance.

Sapd, 2018-06-18:
The problem is, that the current version doesn't support sidetone on G430, because the project changed the communication library from libusb to libhidapi. See https://github.com/Sapd/HeadsetControl/blob/master/src/devices/logitech_g430.c . Someone has to rewrite and test code for libhidapi. 

However, like in the readme stated, you can go to an older version: git reset --hard 41be99379f

VeniVidiViciVersace, 2018-06-18:
I see, sorry about that. I installed that version and got it working. Thank you.

closed 2018-06-18 by VeniVidiViciVersace