Sapd/HeadsetControl #23: volume scroll stops working after launching

alete89, 2018-06-30:
This works before launching HeadsetControl
The headset itself makes a little sound indicating if you are raising or lowering the volume, volume raises or lowers fine and in my case, mate desktop shows this notification:
[image]

After launching `sudo HeadsetControl -s 128`
you can still hear the sound in headset, but volume remains unchanged and no notification is shown, off course.

Eusebius1920, 2018-07-02:
I had the exactly same issue. (Using G930 on Arch Linux with the latest master commit)

Git bisecting showed me that it was introduced in this commit: 46479a49940e9db6bbb69ce2e517bc34e36e4d1b

Further research lead me to this issue of hidapi: https://github.com/signal11/hidapi/issues/305

Solution for me: Using **libhidapi-hidraw** instead of libhidapi-libusb. You can enforce this by applying this git patch (iff your distribution delivers the hidraw version of libhidapi. Otherwise you have to compile it yourself):

```diff
diff --git a/cmake_modules/Findhidapi.cmake b/cmake_modules/Findhidapi.cmake
index 86243fb..ac916e5 100644
--- a/cmake_modules/Findhidapi.cmake
+++ b/cmake_modules/Findhidapi.cmake
@@ -24,7 +24,7 @@
 # http://www.boost.org/LICENSE_1_0.txt)
 
 find_library(HIDAPI_LIBRARY
-	NAMES hidapi hidapi-libusb)
+	NAMES hidapi-hidraw hidapi hidapi-libusb)
 
 find_path(HIDAPI_INCLUDE_DIR
 	NAMES hidapi.h
```

alete89, 2018-07-02:
Thanks, @Eusebius1920 I'll check it out and let you know.

Sapd, 2018-07-06:
Thank you @Eusebius1920 
Actually I thought, that the CMake file uses hidraw, as die libusb implementation isn't a desired one. But I didn't look closely while implementing hidapi. Also I don't quite get, why the libusb implementation doesn't use the auto detach function, which would get rid of the problem.

Hidraw should be on all major distributions, you are welcome to make a pull request. However, is there a particular reason why you still have `hidapi-ibusb` in the find_library directive?

closed 2018-07-06 by Sapd

Eusebius1920, 2018-07-06:
I created a pull request and removed `hidapi-libusb` from the directive. I thought you had some reasons for it in the first place, so i did not delete it.

PR #24 (2018-07-06): Changed CMake to prefer hidapi-hidraw over hidapi-libusb.
Fixes #23