Sapd/HeadsetControl #25: Get the current sidetone level

brunofin, 2018-07-18:
Hello,

Is there a way to get the current sidetone level value for a connected headset? I would like to build a sidetone slider bar for GNOME around your code and would like to know the initial value recorded in the headset itself without needing to persist it on the computer.

Sapd, 2018-10-24:
Hello,
sorry for the late reply.

It definitly needs new captures. I also looked into it and I didn't find any packet for the Corsair Void. I also think, that the Windows-Drivers are saving it persistently (because when you plug the Headset into another Computer, it looses the sidetone setting over time).

Sapd, 2020-05-09:
Info: At least on corsair devices it is definitely saved locally.

closed 2020-05-09 by Sapd