Sapd/HeadsetControl #26: Corsair Void Pro Wireless returns error could not find headset.

Fiiily, 2018-07-23:
Ubuntu 18.04 LTS
[image]

Fiiily, 2018-07-23:
I realized I downloaded a super old version of it right after I posted this and built from source, issue still occours.
[image]

Sapd, 2018-07-23:
Can you post the output of `lsusb -v`?

Fiiily, 2018-07-23:
```Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            4.15
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

Bus 003 Device 002: ID 1b1c:0a1a Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0a1a 
  bcdDevice            0.00
  iManufacturer           3 
  iProduct                4 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
[... 623 lines cut ...]
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```

Sapd, 2018-07-23:
Can you try, opening the file `src/devices/corsair_voidpro.c` and in line 20, change the number to `0x0a1a` and then recompile & retest.

Fiiily, 2018-07-23:
I can not hear myself when I set a sidetone.

Sapd, 2018-07-23:
Try a value like 100

Fiiily, 2018-07-23:
hey, that sounds like me!
108 feels like a sweet spot

Fiiily, 2018-07-23:
Thanks very much for your help, if you need anything more I'll check the issue in a few.

closed 2018-07-23 by Fiiily

Sapd, 2018-07-23:
Sure, I will update the Repo the next days with the wireless version.

nelwa, 2020-07-18:
After following these instructions I get this error:

```
sudo ./headsetcontrol -s 108
Found Corsair Void (Pro)!

ioctl (SFEATURE): Broken pipe
Failed to set sidetone. Error: -1: (null)

```
Any suggestions?
Please