Sapd/HeadsetControl #35: Steelseries Artic 7 2019 not reconnized

olielvewen, 2019-01-28:
Hi,
I've purchased for Christmas an Steelseries Headset Artic 7 version 2019 and it is not reconnized on Linux Manjaro with the last version of HeadsetControl.
`[olivier@thorielle ~]$ HeadsetControl -h
Headsetcontrol written by Sapd (Denis Arnst)
        https://github.com/Sapd

Parameters
  -s level      Sets sidetone, level must be between 0 and 128
  -b            Checks the battery level
  -n soundid    Makes the headset play a notifiation

[olivier@thorielle ~]$ HeadsetControl -b
Headsetcontrol written by Sapd (Denis Arnst)
        https://github.com/Sapd

No supported headset found
[olivier@thorielle ~]$ HeadsetControl -n 0
Headsetcontrol written by Sapd (Denis Arnst)
        https://github.com/Sapd

No supported headset found`

Here more explanation about my system and usb:

`[olivier@thorielle ~]$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 046d:c70a Logitech, Inc. MX5000 Cordless Desktop
Bus 003 Device 003: ID 046d:c70e Logitech, Inc. MX1000 Bluetooth Laser Mouse
Bus 003 Device 002: ID 046d:0b02 Logitech, Inc. C-UV35 [Bluetooth Mini-Receiver] (HID proxy mode)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1b1c:1b51 Corsair 
Bus 001 Device 002: ID 1038:12ad SteelSeries ApS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
====================================================================================================================================
[olivier@thorielle ~]$ dmesg | grep usb
[    0.718696] usbcore: registered new interface driver usbfs
[    0.718696] usbcore: registered new interface driver hub
[    0.718700] usbcore: registered new device driver usb
[    0.869704] usbcore: registered new interface driver usbserial_generic
[    0.869708] usbserial: USB Serial support registered for generic
[    1.718711] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    1.718712] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.718713] usb usb1: Product: xHCI Host Controller
[    1.718714] usb usb1: Manufacturer: Linux 4.19.12-2-MANJARO xhci-hcd
[    1.718714] usb usb1: SerialNumber: 0000:02:00.0
[    1.719509] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.719522] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 4.19
[    1.719523] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.719524] usb usb2: Product: xHCI Host Controller
[    1.719524] usb usb2: Manufacturer: Linux 4.19.12-2-MANJARO xhci-hcd
[    1.719525] usb usb2: SerialNumber: 0000:02:00.0
[    1.720267] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    1.720268] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.720268] usb usb3: Product: xHCI Host Controller
[    1.720269] usb usb3: Manufacturer: Linux 4.19.12-2-MANJARO xhci-hcd
[    1.720270] usb usb3: SerialNumber: 0000:0a:00.3
[    1.720467] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.720480] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 4.19
[    1.720481] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.720481] usb usb4: Product: xHCI Host Controller
[    1.720482] usb usb4: Manufacturer: Linux 4.19.12-2-MANJARO xhci-hcd
[    1.720483] usb usb4: SerialNumber: 0000:0a:00.3
[    2.046734] usb 1-2: new full-speed USB device number 2 using xhci_hcd
[    2.050041] usb 3-2: new full-speed USB device number 2 using xhci_hcd
[    2.224527] usb 3-2: New USB device found, idVendor=046d, idProduct=0b02, bcdDevice=40.02
[    2.224530] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.224531] usb 3-2: Product: Logitech BT Mini-Receiver
[    2.224532] usb 3-2: Manufacturer: Logitech
[    2.360881] usb 1-2: New USB device found, idVendor=1038, idProduct=12ad, bcdDevice= 1.19
[... 43 lines cut ...]
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 10 rev: 2.0 speed: 480 Mb/s 
           chip ID: 1d6b:0002 
           Device-1: 1-2:2 info: SteelSeries ApS type: Audio,HID 
           driver: hid-generic,snd-usb-audio,usbhid interfaces: 6 rev: 1.1 speed: 12 Mb/s 
           chip ID: 1038:12ad 
           Device-2: 1-6:3 info: Corsair type: Mouse,HID driver: hid-generic,usbhid 
           interfaces: 2 rev: 2.0 speed: 12 Mb/s chip ID: 1b1c:1b51 
           serial: 16007025AF099D2258CCD846F5001946 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 4 rev: 3.1 speed: 10 Gb/s 
           chip ID: 1d6b:0003 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 4 rev: 2.0 speed: 480 Mb/s 
           chip ID: 1d6b:0002 
           Hub: 3-2:2 info: Logitech C-UV35 [Bluetooth Mini-Receiver] (HID proxy mode) 
           ports: 3 rev: 2.0 speed: 12 Mb/s chip ID: 046d:0b02 
           Device-3: 3-2.2:3 info: Logitech MX1000 Bluetooth Laser Mouse type: Keyboard 
           driver: hid-generic,usbhid interfaces: 1 rev: 2.0 speed: 12 Mb/s 
           chip ID: 046d:c70e serial: 000761A35EDF 
           Device-4: 3-2.3:4 info: Logitech MX5000 Cordless Desktop type: Mouse 
           driver: hid-generic,usbhid interfaces: 1 rev: 2.0 speed: 12 Mb/s 
           chip ID: 046d:c70a serial: 000761A35EDF 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 4 rev: 3.0 speed: 5 Gb/s 
           chip ID: 1d6b:0003 
[olivier@thorielle ~]$ `

Hope that can help you.

Sapd, 2019-01-28:
Your device has an other id. You could try changing `idProduct` in `HeadsetControl/src/devices/steelseries_arctis7.c` line 15 to 0x12ad

Recompile and try again then.

olielvewen, 2019-02-04:
I've created another package and updated this one was installed thanks the help of French Manjaro forum. 
It will be nice if you could provide a list of products which are supported and not.
Well now my headset is reconized however none features are supported. 
`
# Casque non branché
[olivier@luthien ~]$ HeadsetControl -b
No supported headset found
# Casque branché
[olivier@luthien ~]$ HeadsetControl -b
Found SteelSeries Arctis 7!

Couldn't open device.
# Fonctionnalité non supporté par ce modèle de casque
[olivier@luthien ~]$ sudo HeadsetControl -b
[sudo] Mot de passe de olivier : 
Found SteelSeries Arctis 7!

Error: This headset doesn't support battery status
[olivier@luthien ~]$ 

 `
BTW. As you can see, We get the same output if the device is connected or not.

Sapd, 2019-02-04:
Ah I didn't see that you want to query the battery.
Battery is currently only supported by the Corsair void (pro) as stated under _Other Features_ in the readme

jamesj999, 2019-08-08:
This issue should now be able to be closed. HeadsetControl now supports the 2019 variant of this headset and has battery support for both variants.

closed 2019-08-08 by Sapd

olielvewen, 2019-08-10:
Yes. You have reason.