Sapd/HeadsetControl #36: HS70 Support?

TacoRaptor67, 2019-02-05:
I noticed the Void is supported, so I was wondering.. any chance of the Corsair HS70 being supported? With what I've found this program is my only chance for getting sidetone from this headset on Linux. If it isn't possible I understand, but I thought I'd ask. I personally have little to no experience with Linux otherwise I'd be happy to help with adding support myself.

Sapd, 2019-02-16:
There could be a chance that Corsair uses the same protocol for all their headsets. So you could try opening `src/devices/corsair_voidpro.c` replacing the productid with the of the HS70, recompile and test.

JMmontilla, 2019-05-15:
I just bought an HS70, it would be awesome to add support for it :) (I would like to try the solution proposed by Sapd, but i have no idea about that recompiling thing xd, any help with that?)

Sapd, 2019-12-09:
See https://github.com/Sapd/HeadsetControl/wiki/Adding-a-Corsair-device (it could work)

olback, 2019-12-30:
@TacoRaptor67 @JMmontilla I can upload a version that you can test your HS70 with if you post the output of `lsusb`.

podiki, 2020-07-10:
This works with commit d601fb0 as they have the same device id (I have an HS70). I can fetch the battery level but haven't tried anything else (not sure there is anything else for this headset).

Only thing is that HS70 is missing in the udev rules.

naag, 2020-09-25:
I just tried setting the sidetone level and that works as well as setting the light on / off. By now also the udev rules have been updated, so I just copied those to `/etc/udev/rules.d/` and all is good now.

Just noticed that I have to reactivate the sidetone level after muting & unmuting the mic.

Sapd, 2020-11-02:
fixed in https://github.com/Sapd/HeadsetControl/pull/105

closed 2020-11-02 by Sapd