Sapd/HeadsetControl #41: "no supported headset found"

camr12, 2019-08-08:
Getting "no supported headset found" when trying to connect to my Corsair Void Pro wireless headset. Worked previously, but I've received a new USB controller from Corsair as part of warranty replacement. I've looked at the model number on old and new USB and they are the same.

I also checked the PCI address (?) in dmesg output and I believe it matches the one in the corsair_void_pro.c file. 

Any help would be appreciated.

Sapd, 2019-08-08:
What is the output of `system_profiler SPUSBDataType` (if your on Mac) or `lsusb` if you are on Linux?

camr12, 2019-08-08:
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:c33c Logitech, Inc.
Bus 001 Device 007: ID 1532:005c Razer USA, Ltd
Bus 001 Device 006: ID 1b1c:1b40 Corsair
Bus 001 Device 004: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 003: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 002: ID 1b1c:0a16 Corsair
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

Sapd, 2019-08-08:
The device id seems to have changed. Try to edit `src/devices/corsair_voidpro_wireless.c` and replace the number in line 21 (device_voidpro_wireless.idProduct = 0x0a1a;) with 0a16. Then recompile by simply typing `make` in the build directory again and see if it works.

If it works, we probably have to specify more possible device numbers

camr12, 2019-08-08:
Works. Odd - I swear it matched ids. 

Thanks so much :)

closed 2019-08-08 by camr12

Sapd, 2019-08-08:
So specifically, the id 0x0a16 is missing from source?

camr12, 2019-08-08:
Indeed, following your instructions worked, 0x0a16.

Sapd, 2019-08-08:
Okay, then I will implement something, which allows to specify multiple possible device ids (for one device file).

Sapd, 2019-08-08:
Added in ed5330ceefc28ff1724bcc4f25ca2f99264c7b73