Sapd/HeadsetControl #49: Corsaid Void Wired not working

Deckweiss, 2019-10-18:
I tried normal setup and I tried to follow the instructions from the other closed issue about this, but it didnt work with different errors.

Sometimes it says no supported headset found, one thing I tried found the headset but said broken pipe.

OS: Kubuntu 19.04

My lsusb output is:
Bus 003 Device 007: ID 1b1c:0a18 Corsair 

What can I do to find the problem and make it work?

Sapd, 2019-10-18:
If it shows, no supported headset found (and you added the device id 0a18), then something is wrong with the connection.
Did you also try after a complete restart and other USB ports?

Sapd, 2019-10-31:
This is actually weird because the code does nothing but go through all the devices and look where the ID matched. Probably a good idea would be to set a breakpoint at `src/device_registry.c` for the get_device function or in main.c in the find_device function and see if it is listed by HIDAPI.

But good that it's controllable via ALSA in this case.

closed 2019-12-09 by Sapd