Sapd/HeadsetControl #53: Wrong idProduct for Corsair Void Pro Wireless (RGB)

gitgubbe, 2019-11-13:
Couldn't get HeadsetControl to work as a non-privileged user. Turns out my headset has a different idProduct that the included udev rule in 50-corsair-void-pro.rules:

ID_VENDOR=Corsair
ID_VENDOR_ENC=Corsair
ID_VENDOR_ID=1b1c
ID_MODEL=Corsair_VOID_PRO_Wireless_Gaming_Headset
ID_MODEL_ENC=Corsair\x20VOID\x20PRO\x20Wireless\x20Gaming\x20Headset
**ID_MODEL_ID=0a1a**
ID_REVISION=0000
ID_SERIAL=Corsair_Corsair_VOID_PRO_Wireless_Gaming_Headset
ID_BUS=usb

Changing
SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0a14", MODE="0666"

To
SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0a1a", MODE="0666"

Makes it work as a normal user:

$ HeadsetControl -b
Found Corsair Void (Pro)!

Battery: 61%
$

Sapd, 2019-11-14:
We simply forgot to include newer product ids (when they were implemented) to the udev files. I think we should also put every id of one manufacturer (like corsair) into one file, to make it simpler and to not clutter the users udev folder.

Sapd, 2019-12-09:
So someone who has time and wants to contribute can do this two points:
- Merge udev files of same manufacturer
- Check (and add) product ids into them

Unfortunately I don't have Linux at the moment and therefore I can't do and test it myself (as it's a bit bigger it needs a simple test with a headset).

ShawnCorey, 2019-12-24:
I had the issue where ID_CORSAIR_VOID_RGB_USB in src/devices/corsair_void.c was 0x1b2a and my device was 0x1b23. It was a simple fix, but it looks like there's a bit of a code change to go along with the UDEV rules for some devices.

After changing the product id it works great though. Thanks for making this utility, I'm just sad I didn't find this project much sooner!

Sapd, 2019-12-24:
@ShawnCorey I added the product ids of your to the list in the source code

closed 2020-05-09 by Sapd