Sapd/HeadsetControl #54: save settings?

ProfessorStrawberry, 2019-11-24:
Hi! I love your project. I just would love to save the settings permanently. Is there a way to achieve that? I am using Arch with KDE.

parkerlreed, 2019-11-25:
If you setup udev to allow your user access, you could add the command to the session startup to apply it.

ProfessorStrawberry, 2019-11-25:
I am still a beginner, but I guess I should be able to do that. Thank you

Oh btw.: is it possible to turn of LED, but keep the microphone LED on, when muted. This is the default behavior on the corsair void pro.

Have a good one!

Sapd, 2019-11-26:
> Oh btw.: is it possible to turn of LED, but keep the microphone LED on, when muted. This is the default behavior on the corsair void pro.
> 
> Have a good one!

In theory yes, but it needs further analysing of the protocol to understand how the LED exactly behave (by someone with a corsair headset)

Sapd, 2019-12-09:
I'm closing the issue because it is not something which can be implemented in this repository. But feel free to continue the discussion/question

closed 2019-12-09 by Sapd