Sapd/HeadsetControl #56: Logitech g933-935 Battery issue!

ziadkiwan, 2019-12-10:
Hello,
i have ubuntu 19.10 when i use HeadsetControl -b i get the following:

Found Logitech g933-935 Wireless!

Failed to request battery. Error: -24: (null)

Sapd, 2019-12-11:
Which is your headset? G933 or G935

ziadkiwan, 2019-12-11:
G933, if you want i can give you anydesk access, so you check ?

Sapd, 2019-12-11:
I can't really help with Anydesk, as it could be an implementation issue or some other things.

Does Sidetone work?

ziadkiwan, 2019-12-12:
Hello,

I complied the source, and everything was successful, all the controls fails for some strange reason, if you have any suggestions i would be glad to test or implement!

Sapd, 2019-12-12:
I just looked it up, error 24 is `EMFILE: Too many open files`
I'm not really sure how this can happen. 
You can try increasing the file limit beforehand: `sudo sysctl -w fs.file-max=100000` does this temporarily.

Alternatively, if this does not work, try opening `cmake_modules/Findhidapi.cmake` and in line 27 replace the line by `NAMES hidapi hidapi-libusb)`

I guess there is some kind of bug in a library in your system.

ziadkiwan, 2019-12-13:
Hello, 

You are right! when i did `sudo sysctl -w fs.file-max=100000` things worked, well as you know Linux deals with everything as files, and there is a limit for number of opened files, but I'm 100% sure that i did not open this amount, maybe the library you are relaying on does not have compatibility with Ubuntu 19.10 ?
I will remove the build folder and try editing the `cmake_modules/Findhidapi.cmake` maybe it will be a better permanent fix!,

i will get back to you once it's done!

Thanks for the help!

Sapd, 2019-12-13:
What surprised me is that error 24 is process specified (EMFILE), as error 23 is global (ENFILE) as far as I understand. So the process itself opened up too many.

So either your file limit is ridiculously low (check with `sysctl fs.file-max` after a reboot). Or the hidapi library your distribution ships with, has a bug.

Instead of editing the cmake file (this will change the library to the libusb implementation of hidapi), you can also write the sysctl parameter permanently into /etc/sysctl.conf 
However it would be interesting anyway, how changing the library would affect the bug.

ziadkiwan, 2019-12-19:
Hello Guys, i found the bug that was caused the issue, if you call HeadsetControl -b while it's charging you get this error Failed to request battery. Error: -24: (null).

Sapd, 2019-12-19:
It's probably simply not possible to query the battery while charging. There should be a check in code (e.g. corsair devices simply set a flag).

ziadkiwan, 2019-12-26:
i don't think so, because the on windows you can check the battery while it's charging, anyway it's a great job you are doing i will make sure not to check while charging, because it breaks the code and you have to restart to make it work again!

KrDimitrov, 2020-05-21:
using `headsetcontrol -b` while charging returns 'charging' anyway. 

Implementing battery levels while charging is more difficult, as the current (probably only possible) solution uses the battery voltage to estimate its level. That voltage is different while charging.

fr33-man, 2020-11-03:
Greetings.

 I have G935. Opensuse Tumbleweed has a file limit of `9223372036854775807`. I think HIDAPI needs to be typed with lowercase letters on line 40 at HeadsetControl/cmake_modules/Findhidapi.cmake as that fixed it for me. Can you please test to confirm.

Sapd, 2020-11-04:
Just comment if there are still problems I will reopen it then.

closed 2020-11-04 by Sapd