Sapd/HeadsetControl #59: I can't find the windows download

RandomOofer, 2019-12-26:
Does this even have one? I would really like to not have to install g hub and break my macros to just disable sidetone. :)

Sapd, 2019-12-27:
You can try this
[HeadsetControl-Windows.zip](https://github.com/Sapd/HeadsetControl/files/4005143/HeadsetControl-Windows.zip)

I just compiled it using msys2 and mingw. It runs fine without an headset, but as I don't have a headset here, I can't test it fully. So report back if it works or doesn't.

RandomOofer, 2019-12-28:
I get this output: 
C:\Users\Onni\Downloads\HeadsetControl-Windows>HeadsetControl -s 0
Found Logitech g933-935 Wireless!

Failed to set sidetone. Error: -1: Incorrect function.

KrDimitrov, 2019-12-28:
Same on my windows pc!

Sapd, 2019-12-28:
Could be that the size of the data is not correct.
See https://github.com/signal11/hidapi/issues/135 Windows seems to be more strict.

In the download page of hidapi, there is a hidapi-externals.zip [here](https://github.com/signal11/hidapi/downloads). Download it and run the testgui. Then check what reportID `0x11` expects.

It is also a good idea to check the other features (LED and battery).

This is btw how I compiled:
1. Install http://www.msys2.org and the packages it shows on the site with pacman
2.`pacman -Su`
3.`pacman -S mingw-w64-x86_64-gcc mingw-w64-x86_64-cmake mingw-w64-x86_64-hidapi make git`
4. Open a new window of msys2, when asked, choose mingw 64 bit
5. cd into the project and create a build folder and cd into it
6. `cmake -G"MSYS Makefiles" ..`
7. `make`

RandomOofer, 2019-12-28:
Could not get feature report from device. Error reported was: Incorrect function.

RandomOofer, 2019-12-28:
[image]

Sapd, 2019-12-29:
@RandomOofer 
Try this with the 4 different endpoints in the GUI (connect each one of them beforehand).
For each of them, insert this `0x11,0xff,0x07,0x1a,20` and press Send Feature Report

RandomOofer, 2019-12-29:
Disabling voicemeeter got me further.
046d:0a87
0x11: Incorrect function
0xff: The parameter was incorrect
0x07: The parameter was incorrect
0x1a: The parameter was incorrect
20: The parameter was incorrect

Sapd, 2019-12-29:
Did you insert them one-by-one? You have to insert it as one line besides the button *Send* Feature Report and click this button. But make sure to try the different connections which you can select in the first box.

You can also try this:
`0x11,0xff,0x07,0x1a,20,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0`

RandomOofer, 2019-12-29:
`0x11,0xff,0x07,0x1a,20 `didn't work on any, reporting Incorrect function.
`0x11,0xff,0x07,0x1a,20,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0`, reporting incorrect function.

Sapd, 2019-12-29:
I think I have an idea, can you try this? (It will ask for sidetone level)

[HeadsetHidTest.zip](https://github.com/Sapd/HeadsetControl/files/4008416/HeadsetHidTest.zip)

RandomOofer, 2019-12-29:
Enter wished sidetone level (100 max): 5
Try: 1
HID enumeration failure.

Sapd, 2019-12-29:
Ooops, I specified the wrong vendor/deviceid in code no surprise. Always a bit hard to do something like that remotely.
Binary with the correct deviceid:
[HeadsetHidTest.zip](https://github.com/Sapd/HeadsetControl/files/4008541/HeadsetHidTest.zip)

RandomOofer, 2019-12-29:
IT WORKED!
HeadsetHidTest.exe
Enter wished sidetone level (100 max): 0
Try: 1
        Device - interface: 3 - usage_page: c
        Failed Incorrect function.
Try: 2
        Device - interface: 3 - usage_page: b
        Failed Incorrect function.
Try: 3
        Device - interface: 3 - usage_page: ff43
        success!!
Try: 4
        Device - interface: 3 - usage_page: ff00
        Failed Incorrect function.
Try: 5

Sapd, 2019-12-29:
Ah then I wasn't wrong. For some reason, Windows uses a different endpoint by default. After my tests the interfaceId remains the same and the endpoint can only be distinguished by the variable usage_page. The correct endpoint for my old G930 was ff00. The correct for your device seems to be ff43.
Also I had to adjust the amount of bytes sent to the correct amount, as Windows is very strict here.

According to my tests, there are practically these errors:
1. Incorrect function -> The endpoint is wrong, or the reportid (the first byte in the data) is wrong
2. Wrong parameter -> Wrong amount of bytes sent or wrong bytes

Currently the software does not support searching for *usage_page*, so there is no quick fix. I will probably implement it soon after new year.

RandomOofer, 2020-01-01:
To add, that file you made keeps the sidetone even after restart.

Sapd, 2020-01-02:
I think that's because of Windows.

I now updated the software, you can now try this
[headsetcontrol-win-rel.zip](https://github.com/Sapd/HeadsetControl/files/4016933/headsetcontrol-win-rel.zip)

RandomOofer, 2020-01-02:
Every function works. Also the sidetone settings being kept is lovely. Thank you for being so much assistance. You can ask me later if any new features need testing.

closed 2020-01-02 by RandomOofer

Sapd, 2020-01-02:
Thank you for testing! https://github.com/Sapd/HeadsetControl/pull/62