Sapd/HeadsetControl #63: Please provide more machine-readable interface

andriyor, 2020-01-08:
For example instead of such output
```
➜ HeadsetControl -b                        
Found Logitech G933 Wireless!

Battery: 70%
```
more convenient will be separate this command

```
➜ HeadsetControl -b                        
70
```
model
```
➜ HeadsetControl -m                     
Logitech G933
```
status
```
➜ HeadsetControl -s            
connected
```

Sapd, 2020-01-08:
You can use the `-c` option, which does exactly your first case.

> `HeadsetControl -m`

This is easily implementable.

> HeadsetControl -s 

-s is currently used for sidetone. Also status is in fact not so easy, because we would have to query the headset, whether it is indeed connected.

franciscod, 2020-06-27:
On a similar note, `-c` reads like it's related to the headset's output, not the CLI output!
```
  -s level	Sets sidetone, level must be between 0 and 128
  -b		Checks the battery level
  -n soundid	Makes the headset play a notifiation
  -l 0|1	Switch lights (0 = off, 1 = on)
  -c		Cut unnecessary output 
  -i time	Sets inactive time in minutes, level must be between 0 and 90, 0 disables the feature.
  -m		Retrieves the current chat-mix-dial level setting
```

might be a good idea to use `-q`, usually meaning `quiet`, and maybe put it at the beginning of the list making it clear it's not a headset-related parameter?

Sapd, 2021-11-13:
For compatibility I didn't change -c (which you are right, is a bad choice). However there is now the preferred `--short-output` option

closed 2021-11-13 by Sapd