Sapd/HeadsetControl #66: Logitech G533 Battery Level

sistematico, 2020-01-15:
```
[lucas@majestic ~]:$ sudo headsetcontrol -b
Found Logitech G533!

Failed to request battery. Error: -100: (null)
```
Sidetone works ok.
Any help is welcome.

Sapd, 2020-01-15:
-100 accords to  (we should be probably more specifically tell the error)
```
   //Headset offline
    if (data_read[2] == 0xFF)
        return HSC_ERROR
```

Can you try restarting the headset and plugging usb out and in?

If that does not help, comment the lines 78 and 79 in src/devices/logitech_g533.c out and report back.

sistematico, 2020-01-15:
@Sapd thank for your awesome help.

Un-plugging and plugging the -b works, but if -s is used show me the new error(with the lines commented):

`Failed to request battery. Error: -48: (null)`

I'm using Arch Linux and my HeadSet have 1-day of use

sistematico, 2020-01-15:
[image]

Sapd, 2020-01-15:
What happens when you set `MSG_SIZE` on line 31 to 12 and recompile? (I changed it recently).

Otherwise @giuseppeM99 implemented the device, I can't be of any more help as I don't own it.

sistematico, 2020-01-15:
```
[lucas@majestic build]:$ cmake ..
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found HIDAPI: /usr/lib/libhidapi-hidraw.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lucas/github/HeadsetControl/build
[lucas@majestic build]:$ make 
Scanning dependencies of target headsetcontrol
[ 10%] Building C object CMakeFiles/headsetcontrol.dir/src/main.c.o
[ 20%] Building C object CMakeFiles/headsetcontrol.dir/src/device_registry.c.o
[ 30%] Building C object CMakeFiles/headsetcontrol.dir/src/utility.c.o
[ 40%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/corsair_void.c.o
[ 50%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g930.c.o
[ 60%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g430.c.o
[ 70%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g533.c.o
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c: In function ‘g533_send_sidetone’:
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:100: error: excess elements in array initializer [-Werror]
   35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
      |                                                                                                    ^
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:100: note: (near initialization for ‘data’)
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:103: error: excess elements in array initializer [-Werror]
   35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
      |                                                                                                       ^
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:103: note: (near initialization for ‘data’)
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:106: error: excess elements in array initializer [-Werror]
   35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
      |                                                                                                          ^
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:106: note: (near initialization for ‘data’)
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:109: error: excess elements in array initializer [-Werror]
   35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
      |                                                                                                             ^
/home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:109: note: (near initialization for ‘data’)
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:141: CMakeFiles/headsetcontrol.dir/src/devices/logitech_g533.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:388: CMakeFiles/headsetcontrol.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

sistematico, 2020-01-15:
> What happens when you set `MSG_SIZE` on line 31 to 12 and recompile? (I changed it recently).
> 
> Otherwise @giuseppeM99 implemented the device, I can't be of any more help as I don't own it.

Ok, I try to clone giuseppeM99/HeadsetControl right?

sistematico, 2020-01-15:
Same error, BTW, @giuseppeM99 don't have issues on repo page.

Sapd, 2020-01-15:
> > What happens when you set `MSG_SIZE` on line 31 to 12 and recompile? (I changed it recently).
> > Otherwise @giuseppeM99 implemented the device, I can't be of any more help as I don't own it.
> 
> Ok, I try to clone giuseppeM99/HeadsetControl right?

No, I mean you’d have to wait for an answer of him, if he is aware/has time.

> [lucas@majestic build]:$ cmake ..
> -- The C compiler identification is GNU 9.2.0
> -- The CXX compiler identification is GNU 9.2.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C [...]
 

Some lines below is the data variable, delete the four 0 after bum

sistematico, 2020-01-15:
> > > What happens when you set `MSG_SIZE` on line 31 to 12 and recompile? (I changed it recently).
> > > Otherwise @giuseppeM99 implemented the device, I can't be of any more help as I don't own it.
> > 
> > 
> > Ok, I try to clone giuseppeM99/HeadsetControl right?
> 
> No, I mean you’d have to wait for an answer of him, if he is aware/has time.
> 
> > [lucas@majestic build]:$ cmake ..
> > -- The C compiler identification is GNU 9.2.0
> > -- The CXX compiler identification is GNU 9.2.0
> > -- Check for working C compiler: /usr/bin/cc
> > -- Check for working C compiler: /usr/bin/cc -- works
> > -- Detecting C compiler ABI info
> > -- Detecting C compiler ABI info - done
> > -- Detecting C compile features
> > -- Detecting C compile features - done
> > -- Check for working CXX compiler: /usr/bin/c++
> > -- Check for working CXX compiler: /usr/bin/c++ -- works
> > -- Detecting CXX compiler ABI info
> > -- Detecting CXX compiler ABI info - done
> > -- Detecting CXX compile features
> > -- Detecting CXX compile features - done
> > -- Found HIDAPI: /usr/lib/libhidapi-hidraw.so
> > -- Configuring done
> > -- Generating done
> > -- Build files have been written to: /home/lucas/github/HeadsetControl/build
> > [lucas@majestic build]:$ make
> > Scanning dependencies of target headsetcontrol
> > [ 10%] Building C object CMakeFiles/headsetcontrol.dir/src/main.c.o
> > [ 20%] Building C object CMakeFiles/headsetcontrol.dir/src/device_registry.c.o
> > [ 30%] Building C object CMakeFiles/headsetcontrol.dir/src/utility.c.o
> > [ 40%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/corsair_void.c.o
> > [ 50%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g930.c.o
> > [ 60%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g430.c.o
> > [ 70%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_g533.c.o
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c: In function ‘g533_send_sidetone’:
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35100 error: excess elements in array initializer [-Werror]
> > 35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
> > |                                                                                                    ^
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35100 note: (near initialization for ‘data’)
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:103: error: excess elements in array initializer [-Werror]
> > 35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
> > |                                                                                                       ^
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:103: note: (near initialization for ‘data’)
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:106: error: excess elements in array initializer [-Werror]
> > 35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
> > |                                                                                                          ^
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:106: note: (near initialization for ‘data’)
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:109: error: excess elements in array initializer [-Werror]
> > 35 |     uint8_t data[MSG_SIZE] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };
> > |                                                                                                             ^
> > /home/lucas/github/HeadsetControl/src/devices/logitech_g533.c:35:109: note: (near initialization for ‘data’)
> > cc1: all warnings being treated as errors
> > make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:141: CMakeFiles/headsetcontrol.dir/src/devices/logitech_g533.c.o] Error 1
> > make[1]: *** [CMakeFiles/Makefile2:388: CMakeFiles/headsetcontrol.dir/all] Error 2
> > make: *** [Makefile:141: all] Error 2
> 
> Some lines below is the data variable, delete the four 0 after num

Ah! Ok!!! One sec.

sistematico, 2020-01-15:
Same.
```
[lucas@majestic build]:$ sudo /usr/local/bin/headsetcontrol -b
Found Logitech G533!

Failed to request battery. Error: -48: (null)
```

giuseppeM99, 2020-01-16:
I'll try to recreate this issue and see what i can do

sistematico, 2020-01-17:
> I'll try to recreate this issue, can you tell me more about this?

Ok.

I'm using Arch Linux and brand new Logitech G533 Wireless Headset.

lsusb: https://termbin.com/3j417

I follow the instructions on README of this repo, and instructions from @Sapd:

```
git clone https://github.com/Sapd/HeadsetControl
cd HeadsetControl
.. edit files following the instructions on this thread ..
mkdir build && cd build
cmake ..
make
sudo make install
```
And i got [this](https://github.com/Sapd/HeadsetControl/issues/66#issuecomment-574892073) error.
Before edit i got [this](https://github.com/Sapd/HeadsetControl/issues/66#issue-550447651) error.

Any help is welcome.

sistematico, 2020-01-17:
https://asciinema.org/a/294161

giuseppeM99, 2020-01-17:
ok, i think this is a standby related issue, try to restart your system and headphones and tell me if it works, i'll try to find a better solution to this issue meanwhile

sistematico, 2020-01-17:
> ok, i think this is a standby related issue, try to restart your system and headphones and tell me if it works, i'll try to find a better solution to this issue meanwhile

Same error.

giuseppeM99, 2020-01-18:
try changing on line 67 from
`uint8_t data_request[] = `
to
`uint8_t data_request[MSG_SIZE] = `
recompile and test it
let me know if it works

sistematico, 2020-01-18:
> uint8_t data_request[MSG_SIZE]

Awesome! Thank you!!!

[image]

closed 2020-01-18 by sistematico

giuseppeM99, 2020-01-18:
i'll push a fix asap