Sapd/HeadsetControl #69: Corsair Void Pro USB Can't set sidetone

zosman1, 2020-03-01:
Hello, after compiling the program as explained in the readme i get the following error when i try to set sidetone. Is this a known issue? I was unable to find anything similar.

The output
```sudo ./headsetcontrol -s 120
Found Corsair Void (Pro)!

ioctl (SFEATURE): Broken pipe
Failed to set sidetone. Error: -1: (null)
```

Computer Specifications:
```
- ubuntu 19.10 eoan 
- Corsair VOID PRO Gaming Headset
```

LsUsb:
`Bus 001 Device 009: ID 1b1c:1b5e Corsair Corsair VOID PRO USB Gaming Headset `

strace:
```access("/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.5/1-6.5:1.0/0003:0424:274C.000C/hidraw/hidraw6/uevent", F_OK) = 0
openat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.5/1-6.5:1.0/0003:0424:274C.000C/hidraw/hidraw6/uevent", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
read(3, "MAJOR=239\nMINOR=6\nDEVNAME=hidraw"..., 4096) = 34
read(3, "", 4096)                       = 0
close(3)                                = 0
openat(AT_FDCWD, "/", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
openat(3, "sys", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0555, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "devices", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
openat(3, "pci0000:00", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "0000:00:14.0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
openat(3, "usb1", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "1-6", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
openat(3, "1-6.5", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "1-6.5:1.0", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
openat(3, "0003:0424:274C.000C", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "hidraw", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
close(3)                                = 0
access("/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.5/1-6.5:1.0/0003:0424:274C.000C/hidraw/uevent", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
openat(3, "sys", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0555, st_size=0, ...}) = 0
close(3)                                = 0
openat(4, "devices", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(4)                                = 0
openat(3, "pci0000:00", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
close(3)                                = 0
[... 124 lines cut ...]
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:1872.0003/subsystem", "../../../../../../../bus/hid", 4096) = 28
lstat("/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:1872.0003/uevent", {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:1872.0003/uevent", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
read(3, "DRIVER=hid-generic\nHID_ID=0003:0"..., 4096) = 199
read(3, "", 4096)                       = 0
close(3)                                = 0
uname({sysname="Linux", nodename="-----------", ...}) = 0
openat(AT_FDCWD, "/dev/hidraw0", O_RDWR) = 3
ioctl(3, HIDIOCGRDESCSIZE or HIDIOCGVERSION, 0x7ffe515ab63c) = 0
ioctl(3, HIDIOCGRDESC, 0x7ffe515ab640)  = 0
ioctl(3, HIDIOCSFEATURE(64), 0x7ffe515ac610) = -1 EPIPE (Broken pipe)
dup(2)                                  = 4
fcntl(4, F_GETFL)                       = 0x2 (flags O_RDWR)
fstat(4, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
write(4, "ioctl (SFEATURE): Broken pipe\n", 30ioctl (SFEATURE): Broken pipe
) = 30
close(4)                                = 0
write(2, "Failed to set sidetone. Error: -"..., 42Failed to set sidetone. Error: -1: (null)
) = 42
close(3)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
````

Sapd, 2020-03-04:
Seems to be something local to your system.
Does setting LED work?
Does it behave the same way after plugging it out/in and after a system reboot?

zosman1, 2020-03-04:
> Does setting LED work?

Yes!
However sending notification sound doesn't work.
> Does it behave the same way after plugging it out/in and after a system reboot?

Unfortunately, yes.

Sapd, 2020-03-06:
I'm not really sure about it.
It could be some error in the code or in the hidraw library.

I've read somewhere about issues with the old hidraw library. You can try compiling the new one maintained by libusb, to see if it fixes it for you (uninstall the old one first): https://github.com/libusb/hidapi

narpic, 2020-04-10:
same thing, cant set sidetone, (same error too)
manjaro + Corsair Void Pro Bass RGB.

narpic, 2020-04-13:
yeah just found issue, Corsair Void PRO USB uses audiodg.exe on windows to disable sidetone, em it isn't supported on linux tho, just asked corsair how can I disable it just waiting for response. @zosman1

Sapd, 2020-04-13:
> yeah just found issue, Corsair Void PRO USB (iCUE software) uses audiodg.exe on windows to disable sidetone, em it isn't supported on linux tho, just asked corsair how can I disable it just waiting for response. @zosman1

Maybe you can simply set the sidetone then via alsamixer if on Linux.

narpic, 2020-04-13:
I tried, i was trying to disable all, it only volumed down sound in headphones. so idk really xD

narpic, 2020-04-13:
or maybe im dumb and im choosing not this driver in alsamixer

nelwa, 2020-07-18:
any update on this?  I'm getting the same problem...

ghostyinyi, 2020-07-18:
not possible to disable sidetone
Temat: Re: [Sapd/HeadsetControl] Corsair Void Pro USB Can't set sidetone (#69)Data: 2020-07-18 10:50Nadawca: "Adriaan Nel" Adresat: "Sapd/HeadsetControl" ; DW: "inyi." ; "Comment" ; 
any update on this?  I'm getting the same problem...

nelwa, 2020-07-18:
> not possible to disable sidetone

Ok, so is sidetone not actually supported in all corsair void products as indicated on the main repo page?

ghostyinyi, 2020-07-18:
It depends on model. But this model is not supported to set sidetone.

nelwa, 2020-07-18:
ok, thanks.

zosman1, 2020-07-21:
[Corsair Quick Start Guide](https://www.corsair.com/corsairmedia/sys_master/productcontent/VOID_PRO_USB_QSG_EU.pdf)

[image]

> It depends on model. But this model is not supported to set sidetone.
On windows with the correct drivers it is most definitely something that can be changed. Have you found any documentation to say otherwise?

ghostyinyi, 2020-07-21:
as I said before it doesn't work on linux due of missing audiodg.exe that has windows (it handles audio thinks i think)

smfuller, 2021-05-29:
I'm not sure if this is specifically a problem with the code here.

I was able to build this as of today in Arch Linux for my Void Elite and can adjust the sidetone with `headsetcontrol -s` without issue.

The comments regarding audiodg.exe and Windows don't seem to have any bearing in this case.

alecrm, 2021-08-03:
I'm incredibly new to playing around with usb information (let alone programming in C), but from what I was able to get out of the code with some debugging, it seems like the information is correct up until hid_send_feature_report() inside void_send_sidetone() in the corsair_void.c class. As far as I can tell, the hid_path used to set the hid_device correctly points to my void pro usb headset. I don't know how to print out the value of the hid_device->device_handle or what that would exactly tell me, but that was the only unknown up until the void_send_sidetone() method. Once inside there, I don't know at all what the values of the data array mean, but assuming that's correct, then the only other point of failure from my understanding would be inside the hidapi code once it hits the hid_send_feature_report() method. Wonder if maybe there's some issue on that end?

Sapd, 2021-11-13:
@alecrm Sounds like a Windows Option. The new version which will be released today contains a fix

closed 2021-11-13 by Sapd