Sapd/HeadsetControl #70: Arctis 7 battery incorrect

baconsalad, 2020-03-05:
I have an Arctis 7 2019 edition

`Bus 002 Device 006: ID 1038:12ad SteelSeries ApS`

The battery is incorrect. After charging it all night the battery strength is reported as 64%. This is a pretty new headset so I doubt the battery has degraded that far yet.

```
$ ./headsetcontrol -b
Found SteelSeries Arctis (7/Pro)!

Battery: 64%
```

Sapd, 2020-03-06:
For the battery, it reads directly a register from the usb protocol. So it simply directly reports what the headset reports. Note that the battery reporting in most headsets will always be inaccurate (I would guess around 20% variation).

Did you have a chance to test it on Windows?

It could also be, that for your (ID_ARCTIS_7_2019), we would have to adjust the value. If you find out that 32% (very roughly estimated) is ca. indeed half of the full battery level.

trevarj, 2020-04-01:
I have an Arctis 7 2019 - I have written my own controller in Python (Windows works too) and found that the battery is not a percentage, but it is max 0x75 (117 or around there). Also I noticed that when you send the request for the percentage, the first read does not always pick up the updated value...It's stupid because the way you request the battery is a control transfer and then you have to read from the register after an interrupt fires, instead of a control read. Idk much about USB though to know if this is standard or not.

closed 2023-07-26 by github-actions[bot]