Sapd/HeadsetControl #71: Corsair virtuoso

IceCupe123, 2020-03-07:
cant get the corsair virtuoso to work added it like the other corsair headsets

nkconnor, 2020-03-30:
I downloaded the iCUE software on my Macbook but it seems broken. Also some complaints from other users on the Corsair forums the past month about this. If anybody has a Windows PC, that would be helpful..

Sapd, 2020-03-30:
I need more information, like what OS you have, what you have tried (with sudo/without etc.), and what is the error message.

However it could be, that the virtuso has a different protocol. That would mean that it's protocol also has to be analyzed, like with the other devies.

nkconnor, 2020-03-31:
@Sapd I'm running Ubuntu 18.04. I added the Virutoso device to `corsair_void.c` following the Wiki instructions. Using `sudo`, I can get most of the commands to return:

```
Found Corsair Void (Pro)!

Success!
```

The battery level command hangs indefinitely. No commands have any observable effect on the headset, so I figured probably different protocol.

Curiously, the device ID changed between when I looked at it yesterday and today (it has been unplugged/replugged into USB).

The iCUE software doesn't work on my Mac... so I'm not sure how to move forward with this.

nkconnor, 2020-04-08:
@Sapd I managed to get iCUE working when connected directly to the headset over USB. However, I'm having trouble making sense of the USB traffic and finding the pattern we need.

Using wireshark,

There's two URB CONTROL packets of length 40(submit) and 34(complete) that get sent out to the device everytime sidetone changes. But nothing changes between the packets except for the IO ID.

There's a lot of passive background traffic with byte lengths 32, 88, 288, and 96, but I can't really tell if these are related to the sidetone or not. 

The 96 packet has 64 bytes of 'leftover capture data' which matches the message size I think you are sending with `hid_send_feature_report`. But it says URB_INTERRUPT (completed) and matches a request with length 32.

I'm not sure how to go any further here. Any tips for me?

Sapd, 2020-04-13:
@nkconnor When I tried it in the past with Wireshark I had the issue of not seeing the packet payload. You have probably the same problem.
Try it with usblyzer https://github.com/Sapd/HeadsetControl/wiki/Development

I think would have to look again into getting capturing working with wireshark and writing a wiki entry some time. There are probably scripts for capturing HID devices.

nkconnor, 2020-04-15:
I think usblyzer only supports Windows

Sapd, 2020-04-15:
Right, I forgot about that.

Okay, I just tried it again in Wireshark and it was no problem at all (probably because I have much more experience now than when I first wrote the software).
However I did it in a windows VM, as I didn't want to disable SIP for Wireshark on my Mac.

[image]
[image]

When you look in the data, you can clearly see some change. 

If you don't see any value change, make sure the sidetone actually changes (so don't keep your headset just on the table). Because I noticed the same problem, and then also noticed that the headset simply bugged.

For filtering I used these arguments
`usb.bInterfaceClass == HID && usb.endpoint_address.direction != IN && usb.src == host && usb.setup.wLength == 64`
Note that I added a length, as I noticed that a lot of keep-alive (or some other noise) packages were sent with a different length.

You can also see, that the data in the image is highly similar to the implementation in corsair_void.c:
`unsigned char data[64] = { 0xFF, 0x0B, 0, 0xFF, 0x04, 0x0E, 0xFF, 0x05, 0x01, 0x04, 0x00, num, 0, 0, 0, 0 };`
(Just the 6-th byte, counting from 0, is in my capture a 0x01 instead of 0xFF is different. Probably software update or some feature activated?)

nkconnor, 2020-04-19:
Thanks for getting this far in here. Very strange, I get really good info from Wireshark when I turn sidetone or other features on/off, but really can't see anything meaningful when adjusting the value/slider.

Possibly SIP or something else I'm doing is interfering..

Sapd, 2020-04-19:
> Thanks for getting this far in here. Very strange, I get really good info from Wireshark when I turn sidetone or other features on/off, but really can't see anything meaningful when adjusting the value/slider.
> 
> Possibly SIP or something else I'm doing is interfering..

Did you see a SET_REPORT packet with length 100 (decimal; 64 in hex)?
Also try the mute button. I noticed when trying to mute/unmute, it will send the sidetone packet first, too (for some reason with sidetone set when mute activated, and unset/set 0 if unmuted).

nkconnor, 2020-04-19:
I think maybe found the issue. All recorded traffic has `usb.dst == "host"`.. there is no recorded traffic with `usb.src == "host"` (like your filter shows)

Sapd, 2020-04-19:
The Wireshark wiki says, that you have to deactivate SIP if you have Catalina. Probably it's this.
Otherwise I guess you'll have to use Virtualbox or some other VM solution for capturing.

themoosman, 2020-05-07:
I have this headset and a working test Win VM.  How can I be of assistance?

Sapd, 2020-05-07:
@themoosman You can find a tutorial here https://github.com/Sapd/HeadsetControl/wiki/Development

joseant79, 2020-05-09:
I have a Virtuoso too, this is what I was able to get
Side-tone full up:
[image]

Side-tone all the way down: 

[image]

Sapd, 2020-05-09:
Try to select only the HID-devices on the left side.

Usually it should be a report. So check those packets, too.

joseant79, 2020-05-11:
Alsamixer is able to control the sidetone correctly on the Virtuoso, the same way as the iCue software does, now the only thing that I am interested  is in check the remaining battery

joseant79, 2020-05-11:
BTW, I tried selecting only the HID devices and still showing nothing, don't know if I am doing something wrong

nfaggian, 2020-07-06:
@joseant79 : what does your alsamixer output look like?

nkconnor, 2020-07-06:
Forgot to subscribe here. Also curious how you are able to get this to work in alsamixer @joseant79

nkconnor, 2020-07-06:
[image]

It works for me, too! I had to make sure the multi-channel input selection was made and then adjust the first 'Headset' in alsa. There's a patch to rename that to 'Sidetone' pending I think.

vuurvoske, 2020-07-07:
Hi, i just came here to tell that i have my virtuoso se working on ubuntu 20.04 out of the box. Issues i noticed:
- low volume (in comparisement to windows 10), firefox (youtube) however is extremely loud in comparisement
- very annoying sidetone.
- The slightest thing triggers the headset to enable again.

nkconnor, 2020-07-07:
@vuurvoske you probably have multiple volume channels causing the volume discrepancy. Check Ubuntu sound settings

joseant79, 2020-07-08:
[image]

It was working well out of the box, just cranked the sidetone up and then press the mic button for 5 seconds and voila!

nfaggian, 2020-08-21:
Anyone had any luck with these headphones and turning off the LED's in linux?

iommu, 2020-09-19:
Ok looking at the wireshark data it seems like the data is quite weird having the first 2 and last 2 bytes swapped as well as having a truncated 0x1XXXX as the 5th byte when the value is 100% so it starts at 0x10400 and goes down to 0xd600 represented at 0x0004 and 0x00d6 respectively. 

However as others have shown the sidetone does work in alasmixer and as such the alsa implementation code is mirrored on github [here](https://github.com/torvalds/linux/blob/c9c9735c46f589b9877b7fc00c89ef1b61a31e18/sound/usb/mixer_maps.c#L357) so it might be easier to reconsturct the alsa code that deconstruct the iCUE USB stream

hmajid2301, 2020-12-31:
Any way to get the battery level in Ubuntu ?

AR2000AR, 2021-04-30:
> @Sapd I'm running Ubuntu 18.04. I added the Virutoso device to `corsair_void.c` following the Wiki instructions. Using `sudo`, I can get most of the commands to return:
> 
> ```
> Found Corsair Void (Pro)!
> 
> Success!
> ```
> 
> The battery level command hangs indefinitely. No commands have any observable effect on the headset, so I figured probably different protocol.
> 
> Curiously, the device ID changed between when I looked at it yesterday and today. Possibly this is due to a firmware update from messing around with the iCUE mac software. The iCUE software barely works, so I'm not sure how to move forward with this in absence of a PC.

In my case the command `headsetcontrol -b` return `Battery: Unavailable` when I change the volume form the headset.

RichardsonWTR, 2022-02-09:
Does this help? https://github.com/ckb-next/ckb-next/issues/785

Sapd, 2022-02-09:
> Does this help? [ckb-next/ckb-next#785](https://github.com/ckb-next/ckb-next/issues/785)

The report itself not. Someone needs to take a look at it with WireShark https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-new-headset
for turning off LED or getting the Battery level.
(Sidetone can not be supported as outlined some posts above, as it is controlled by an audio/volume channel).

Patola, 2022-09-19:
So the Corsair Virtuoso support is not yet done on this application? I'd have to edit the source code to get it working?

For some reason pipewire did not show the 'sidetone' slider here. What should I do for it to show?

jayofdoom, 2022-09-19:
Yeah, as far as I'm aware this headset is still not working wirelessly on Linux.

Patola, 2022-09-19:
Thanks, @jayofdoom. Just a slight correction on your wording, the headset settings (sidetone, RGB settings) still are not working wirelessly on Linux, but the headset works fine otherwise -- you just can't adjust anything other than the volume.

jayofdoom, 2022-09-19:
Even with the wireless dongle? I didn't realize that had been fixed. I'll try to make it work. I thought it had to be directly connected with a cable.

Patola, 2022-09-20:
Yes, even with the wireless dongle.

AR2000AR, 2022-09-20:
Using it wirelessly too. Never had any issue (debian 10,11 and popos)

Sapd, 2023-05-01:
Closing, the sidetone is controlled via an audio channel for Virtuso, which we can't access via HID.
However feel free to open another issue for other features like battery.

closed 2023-05-01 by Sapd