Sapd/HeadsetControl #73: Could't open device.

KannstGarnichts, 2020-03-26:
~/HeadsetControl/build$ ./headsetcontrol -s 16
Found Logitech G633/G933/G935!

Couldn't open device.

Any solutions?

Sapd, 2020-03-26:
Probably no permissions. Use sudo or install the respective udev rules by typing `make install` as stated in the README.

hosler, 2020-03-27:
I needed to add my VOID product id to the udev rules: 0a1a

# My VOID
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0a1a", TAG+="uaccess"

matzedings, 2020-04-01:
Can confirm that permissions are not working even with `sudo make install`, I still need to use sudo.

If this is really due to udev rules, could you implement this into your code?

morhook, 2021-09-21:
Had same issue as reporter! Fixed the same way. Thanks a lot! (I think we should close this issue)

closed 2021-11-13 by Sapd