Sapd/HeadsetControl #82: Gentoo ebuild

SabbathHex, 2020-04-30:
I wrote a Gentoo ebuild for headsetcontrol. Since there is no installation section in the repo - should I add it as a wiki page?

Sapd, 2020-04-30:
Yes, if it is not a one/two liner, you can add it there; and then link it via the Building section - with the info that it it possible to use ebuild instead of building yourself

SabbathHex, 2020-04-30:
Thanks for the reply. The ebuild applies a patch to the udev rules location and reloads the udev rules after the package is installed. 

Looks like the wiki is closed for editing:
[image]

For an editable wiki:
[image]

Sapd, 2020-04-30:
Ahh, try again, I changed the permissions.

SabbathHex, 2020-04-30:
Works, thanks! I'll submit the PR
[image]

closed 2020-04-30 by SabbathHex