Sapd/HeadsetControl #86: Arch Linux AUR Package

sehnem, 2020-05-02:
Hey, congratulations for the great work!

Just came here to inform that I created and AUR package for this repository [headsetcontrol-git](https://aur.archlinux.org/packages/headsetcontrol-git/). Anyone using Arch based distributions have an easier way to install it now.

Sapd, 2020-05-02:
Thank you :)

Just out of curiosity, what is the difference between the package you created and https://aur.archlinux.org/packages/headsetcontrol ?

sehnem, 2020-05-02:
Ohh, there was a package? sorry, I did not see in the readme and duplicated it, I think it is the same package.

closed 2020-05-02 by sehnem

parkerlreed, 2020-05-03:
Technically yours was more correct. git packages need to have -git on them. Regular are for pointy releases.

https://wiki.archlinux.org/index.php/VCS_package_guidelines