Sapd/HeadsetControl #88: Enhancement: Future plans?

Thalass, 2020-05-07:
Hi! I just bought a G935, and HeadsetControl has been exactly what I needed to get the headset working. Thanks! 
Are there any plans to expand the features to cover the other things that the official software can do? I don't have a windows machine, and right now the buttons are useless to me. 

Many thanks.

Sapd, 2020-05-07:
Do you mean macro-buttons? I'm not quite sure, but I think this is possible with a x11 configuration on Linux (probably something like this https://askubuntu.com/questions/152297/how-to-configure-extra-buttons-in-logitech-mouse )

I myself don't have specific further plans for G930/G935, as I don't use it any more. However everyone can contribute via pull requests.

Thalass, 2020-05-07:
Thanks for that. I didn't find that before. Probably too narrow a search.

KrDimitrov, 2020-05-21:
Did it work?

There is a way to enable the G-buttons using hidapi. The headset then sends data when a button is pressed. Problem: The microphone wont mute/unmute when the gbuttons are enabled. Probably because its controlled by software and i couldnt find a fix for that. 

There's also a way to monitor when the microphone is muted/unmuted, just like with the buttons. On windows its on a different hid path though and i think that would require opening multiple devices.

Sapd, 2020-05-24:
> There is a way to enable the G-buttons using hidapi. The headset then sends data when a button is pressed. Problem: The microphone wont mute/unmute when the gbuttons are enabled. Probably because its controlled by software and i couldnt find a fix for that.

It's the same on corsair. There you have to poll for the mute button and then mute the headset via another hid_write. However this makes the software more complex.

closed 2021-11-13 by Sapd