Sapd/HeadsetControl #92: Notification daemon

Manawyrm, 2020-05-22:
Hi! 

[image]

Thank you very very much for this useful piece of software. 
I wrote a small (and pretty ugly) daemon that will display notifications for the battery state of your devices: 
https://github.com/Manawyrm/headsetcontrol-notificationd

Should we link to this repo in the README for HeadsetControl? 
I'm probably not the only user that would like to have desktop notifications for battery status. 

Thanks!

centic9, 2020-05-23:
In the same vein, if you add a "related tools" section there is another tool based on HeadsetControl: 

I wrote a small app-indicator which displays the current battery percentage and provides a menu to control features of the headset (lights, sidetone, ...). It uses HeadsetControl to do the actual communication with the headset

It's available at https://github.com/centic9/headset-charge-indicator/

[image]

Sapd, 2020-05-24:
Sure, look here https://github.com/Sapd/HeadsetControl/pull/93/files

Does the description suit you both?

Manawyrm, 2020-05-24:
Cool, fine by me :)

Manawyrm, 2020-06-02:
Just a small nitpick: 
There's a typo in the Readme, it says "headsetcontrol-notifcationid" instead of "headsetcontrol-notifcationd". 
Should I send a PR to fix that or do you want to just change it?

Sapd, 2020-06-06:
I quickly fixed it

closed 2020-06-06 by Sapd