Sapd/HeadsetControl #103: Corsair Void Elite Surround 7.1 identifies as another Corsair headset

stefan-vspec, 2020-08-27:
I was able to go a git pull and copy the repository locally to my machine, but then had to go and edit the /HeadsetControl/src/devices/corsair_void.c file in order to add my device

To be specific, I have a Corsair Void Elite Surround with a 7.1mm jack for surround sound capabilities

I was able to add the device through the constant

#define ID_CORSAIR_VOID_ELITE_SURROUND_USB

doing lsusb gave back >> Bus 001 Device 010: ID 1b1c:0a57 Corsair Corsair VOID ELITE Surround USB Adapter

Following the supplemental tutorial in the README section of the primary repository, my device ID is 0a57

This is the src code after i finished editing it:

include <string.h>

static struct device device_void;

enum void_wireless_battery_flags {
    VOID_BATTERY_MICUP = 128
};
/*
#define ID_CORSAIR_VOID_WIRELESS           0x1b27
#define ID_CORSAIR_VOID_PRO                0x0a14
#define ID_CORSAIR_VOID_PRO_R2             0x0a16
#define ID_CORSAIR_VOID_PRO_USB            0x0a17
#define ID_CORSAIR_VOID_PRO_WIRELESS       0x0a1a
#define ID_CORSAIR_VOID_RGB_USB            0x1b2a
#define ID_CORSAIR_VOID_RGB_USB_2          0x1b23
#define ID_CORSAIR_VOID_RGB_WIRED          0x1b1c
#define ID_CORSAIR_VOID_ELITE_WIRELESS     0x0a55
#define ID_CORSAIR_VOID_RGB_ELITE_WIRELESS 0x0a51
#define ID_CORSAIR_VOID_ELITE_USB          0x0a52
#define ID_CORSAIR_HS70_PRO                0x0a4f
*/
#define ID_CORSAIR_VOID_ELITE_SURROUND_USB 0x0a57

static const uint16_t PRODUCT_IDS[] = {
/*
    ID_CORSAIR_VOID_RGB_WIRED,
    ID_CORSAIR_VOID_WIRELESS,
    ID_CORSAIR_VOID_PRO,
    ID_CORSAIR_VOID_PRO_R2,
    ID_CORSAIR_VOID_PRO_USB,
    ID_CORSAIR_VOID_PRO_WIRELESS,
    ID_CORSAIR_VOID_RGB_USB,
    ID_CORSAIR_VOID_RGB_USB_2,
    ID_CORSAIR_VOID_ELITE_WIRELESS,
    ID_CORSAIR_VOID_RGB_ELITE_WIRELESS,
    ID_CORSAIR_VOID_ELITE_USB,
    ID_CORSAIR_HS70_PRO,
*/
    ID_CORSAIR_VOID_ELITE_SURROUND_USB,
};

I commented out most of the devices so there wouldnt be any errors in which Name/ID pair was referenced by the function used to find the devices

I then go to /HeadsetControl/build and make install the new file

This is successful:

Scanning dependencies of target headsetcontrol
[  8%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/corsair_void.c.o
[ 16%] Linking C executable headsetcontrol
[100%] Built target headsetcontrol
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/bin/headsetcontrol
-- Up-to-date: /etc/udev/rules.d
-- Up-to-date: /etc/udev/rules.d/70-headsets.rules

and then i run headsetcontrol -s 0 to set the sidetone to 0.

This returns: 
Found Corsair Void (Pro)!

Couldn't open device.

...weird....I dont have a Corsair Void (Pro)

So either Corsairs ID protocol changed, but that makes no sense because there are commits to this repo from 6 days ago, which makes this code recent

So my issue is that HeadsetControl is identifiying my Corsair Void Elite Surround as a Corsair Void Pro

Any ideas on how i can fix this?

Sapd, 2020-08-27:
> Couldn't open device.

This means probably that it doesn't have permissions, did you try running it with sudo? To run it without sudo, you also have to edit the file in the udev folder.

> So my issue is that HeadsetControl is identifiying my Corsair Void Elite Surround as a Corsair Void Pro
> 
> Any ideas on how i can fix this?

The string the software returns, is for all devices in the file you edited the same, it is simply hard coded. So it does indeed detect the correct one. 
You also don't have to comment out other code or other IDs. They simply will be iterated by the software to see, wether one of them matched one plugged in device, if so, it will use the code in the file, which is the same for all devices which are also listed in that file.

stefan-vspec, 2020-09-01:
So I tried running the program with sudo, and now it returns:

(I ran $ sudo headsetcontrol -s 0)

Found Corsair Void (Pro)!

ioctl (SFEATURE): Broken pipe
Failed to set sidetone. Error: -1: (null)

As for setting the file to run without sudo, how do I edit the 70-headset.rules in udev/ to accept me as a co-owner 
Can I not just use chown -R to give me full ownership of the folder?

Thanks :D

closed 2020-09-01 by stefan-vspec