Sapd/HeadsetControl #104: Feature request : Add RGB led support

iommu, 2020-09-19:
Add a parameter for changing the color of RGB LED capable headsets, alternatively extend the functionality of the "-l" parameter to accept hex codes or similar as well as the 0/1 for turning on/off lights

Sapd, 2020-09-19:
Not easily possible.

The headsets have two kind of different operating modes:
- Hardware Mode
    - Mute Button controlled by headset
    - Headset accepts sidetone commands and saves them until shut down
    - Headset does not accept RGB commands, and sets the LEDs to a default value (like red solid on Corsair)
- Software Mode
    - Mute Button is just a button, like a normal macro button. The software needs to send the headset back, that it should really mute (on every button press)
    - After every mute, Sidetone must be adjusted again
    - Headset may receive a stream of LED settings/values. E.g. a fading effect creates a lot of USB traffic, as the software itself needs to update the LEDs every x milliseconds 

So sending LED commands is only possible, by using some kind of daemon. As otherwise other functions like Mute wouldn't work anymore.

For Corsair, I did a draft of that functionality, but it needs a lot of further work. Also I'm not sure if this is the right way from a software architecture point of view: https://github.com/Sapd/HeadsetControl/issues/57

iommu, 2020-09-19:
Hmm, so this would better be handled by something like ckb-next (at least for my corsair virtuoso) which does operate using a daemon

Sapd, 2021-11-13:
Currently out of scope. But feel free to open up a discussion ( https://github.com/Sapd/HeadsetControl/discussions )

closed 2021-11-13 by Sapd