Sapd/HeadsetControl #108: Logitech G733 support

Froghut, 2020-10-23:
I recently got a G733 and successfully managed to find the sidetone setting thanks to the tutorial in the wiki, but I have no idea how to get the product id and how to find the message needed to send to get battery level/voltage. I can provide Wireshark logs etc., if anyone would be able to help me I would greatly appreciate it!

Sapd, 2020-10-23:
Hello,

you can read this article https://github.com/Sapd/HeadsetControl/wiki/Development it describes how to finde the messages with Wireshark. 

Feel free to ask additional questions.

Froghut, 2020-10-23:
Well I used that article to find the sidetone value, but there is no info there on how to find the product id and voltage/battery level. Any pointers in that regard would be appreciated.

Sapd, 2020-10-23:
Ah now I understand, I was not sure if you meant this article, getting the PID ist easy, just look into the device manager (see https://www.synology.com/en-us/knowledgebase/DSM/tutorial/Compatibility_Peripherals/How_do_I_check_the_PID_VID_of_my_USB_device ).

Battery info is indeed a bit more difficult. The best method is to filter all packets out - which apperently dont contain battery info - with wireshark rules(the "background noise"). Then let the headset discharge, and look for packets with a slowly decreasing number. Depending on the amount of packets, this could probably take some time.

Froghut, 2020-10-23:
Great, thanks! I will try my luck :)

Froghut, 2020-10-23:
I tried to add support for g733 (sidetone only for now) just to get started, but I get the error "Incorrect function". I made sure to use the correct product id and vendor id, I also checked the usage id + page using the HIDAPI tester tool and sending the message using that tool I can successfully control the sidetone. If anyone could have a look at my code I would be grateful. I forked the project and commited my changes to a branch: https://github.com/Froghut/HeadsetControl/tree/G733 - not sure if this is the "correct" way to do this, I have barely used github so I hope this is ok.

Sapd, 2020-10-26:
> I tried to add support for g733 (sidetone only for now) just to get started, but I get the error "Incorrect function". I made sure to use the correct product id and vendor id, I also checked the [...]

Maybe little-endian, big-endian issue (bytes are shown the other way around in the software for productid) - but not sure

> Edit2: Is it possible to find out which prosuct id is "used" when the "request_battery" function gets called? it appears the G733 voltage is different from the other headsets, so I would need to calculate this differently from the other devices.
> Edit3: It looks like the logitech software is somehow showing the wrong percentage and the voltage is the same as the other devices. I will open a pull request.

The battery measurement in the headsets is very very inprecise. It just uses voltage measurement which is inaccurate (instead of measuring the current and adding it up over time). So voltage and real battery level can even differ between two headsets of the same type.

gaetancollaud, 2021-01-19:
Seems to be fixed by #121

At least for me it worked out of the box. The readme doesn't mention `G733` but I had no issue. Even for change the led state.

I first had the issue with root permission. This is why I though G733 was not supported.

shanec42, 2021-01-28:
Confirm, this works on my G733 as long as I run headsetcontrol as root.

Thanks!

closed 2021-11-13 by Sapd