Sapd/HeadsetControl #115: Adding Support for Steelseries Arctis 9

JohnRoux, 2020-11-20:
Would it be possible to add support for the Arctis 9?

Fundamentally it should be almost the exact same as the Arctis 7's

Sapd, 2020-11-21:
You can try just adding the product id of your headset to the current implementation. Similar to https://github.com/Sapd/HeadsetControl/wiki/Adding-a-Corsair-device

But ofc in this file then: https://github.com/Sapd/HeadsetControl/blob/master/src/devices/steelseries_arctis_7.c

JohnRoux, 2020-11-23:
Tried that :(

Sadly it doesn't actually seem to be the same as the Arctis 7's

It has 2 devices when running `lsusb` :
```
Bus 001 Device 013: ID 1038:12c2 SteelSeries ApS
Bus 001 Device 014: ID 1038:12c4 SteelSeries ApS
```

When adding the 12c2, it gave this error when doing either battery or mix checks:
```
Failed to request battery. Error: -1: (null)
```

When adding the 12c4, it gave this error when doing either battery or mix checks:
```
Requested/supported HID device not found or system error.
```

When adding both, it gave the 12c2 error

dark, 2020-11-24:
John, I think you might have mixed up the error messages.
I am trying to make it work as well, so adding extra info here, if it can be helpful.

I have run `lsusb -v` on both devices (full output below).
At a glance, the product ID `:12c4` seems to be the actual headset, and `:12c2` the wireless dongle, perhaps? I have not confirmed that yet.

I tried to query for the battery on interface 5 of `:12c4` and the `hid_write` in `arctis_7_request_battery` fails (hence the error "Error: This headset doesn't support battery status").
I have also tried interfaces 0 through 2 of `:12c2`. Interfaces 0 and 2 hang on `hid_read` in `arctis_7_request_battery`, while interface 1 fails on `hid_write` (like the interface for the other device did).

I will do some more debugging in the next few days, including more queries. Please let me know if more information is needed in the meantime, I might be able to get to it sooner.

```
Bus 003 Device 007: ID 1038:12c2 SteelSeries ApS 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1038 SteelSeries ApS
  idProduct          0x12c2 
  bcdDevice            0.71
  iManufacturer           1 SteelSeries
  iProduct                2 SteelSeries Arctis 9
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 415 lines cut ...]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      26
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

centic9, 2020-11-24:
Based on changes committed at https://github.com/atagulalan/arctis-battery-percentage/compare/3989f67530...3619505dd8 it seems produc
    reportIdx: 3,
    writeBytes: [0x06, 0x12],

centic9, 2020-11-24:
Also there is some documentation about analyzing the protocol at https://github.com/Sapd/HeadsetControl/wiki/Development, maybe some other features can be added as well this way.

pdefert, 2020-11-24:
under windows, arctis 9  is seen as 2 headset (one for gaming, one for chat ) .
under linux, lsub see 2 usb devices but only one is considerate as audio device (12c4 for me)

i modify steelseries_arctis_7.c as proposed. And i modifed 70-headsets.rules as follow:
>#SteelSeries ARCTIS 9 CHAT
>KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1038", ATTRS{idProduct}=="12C4", TAG+="uaccess"
>#SteelSeries ARCTIS 9 GAMING
>KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1038", ATTRS{idProduct}=="12C2", TAG+="uaccess"

- with device 12c4, i have  
> Found SteelSeries Arctis (7/9/Pro)!
> Couldn't open device.
- with device 12c2 
>Found SteelSeries Arctis (7/9/Pro)!
>Requested/supported HID device not found or system error.

Sapd, 2020-11-24:
The Udev files are just for permissions.

You will have to analyze the protocol. I also don’t think that further debugging will help.  Like @centic9 said, the wiki has a step by step tutorial.

Tubbles, 2020-12-29:
So I bought a SteelSeries Arctis 9 today and found out that it did not work out of the box. Any progress on this? :)

rhwilr, 2020-12-30:
I also bought an Arctis 9 a few days ago and have been working on adding support to HeadsetControl. 
I have sidetone and inactive-time working, and am currently working on battery. It seems to be quiet a bit different from the other models. 

Once I figure it out I'll submit a PR.

Tubbles, 2020-12-30:
> I also bought an Arctis 9 a few days ago and have been working on adding support to HeadsetControl. 
> I have sidetone and inactive-time working, and am currently working on battery. It seems to be quiet a bit different from the other models. 
> 
> Once I figure it out I'll submit a PR.

Well I haven't even been able to get mine to register an audio channel or work at all, might I ask for your setup of sound stack and version?

Sapd, 2021-01-06:
Fixed by #126

closed 2021-01-06 by Sapd