Sapd/HeadsetControl #119: Doesn't detect on arch

headassbtw, 2020-12-15:
Recently reinstalled arch(manjaro) and noticed that the program can't find my headset anymore. it's a corsair void elite RGB wireless (which IS supported), but the program can't find it. it shows under lsusb as well, not sure what the problem is.
any help i can give or get is welcome
[image]

Sapd, 2020-12-15:
hmm maybe you pulled an old version. I just checked and the id of your headset is in the source code.

How did you install HeadsetControl?

headassbtw, 2020-12-15:
um, pamac, my graphical package manger of choice, was that the problem? do i need to compile from source?

parkerlreed, 2020-12-16:
Install headsetcontrol-git from AUR

headassbtw, 2020-12-16:
sick, that worked, thank you!

closed 2020-12-16 by headassbtw