Sapd/HeadsetControl #124: Can't compile using latest Cmake (3.19.2-1)

vitor180396, 2020-12-31:
[image]

Sapd, 2020-12-31:
Is the *make* package installed?

vitor180396, 2021-01-01:
> Is the _make_ package installed?

Yes
[image]

core1024, 2021-01-01:
You're obviously using Arch Linux based distro. Have you tried installing it from AUR? Or at least check the PKGBUILD. There are listed all the dependencies you need. If you have `yay` installed you only need to issue `yay -S headsetcontrol-git` as normal user.

vitor180396, 2021-01-02:
> 
> 
> You're obviously using Arch Linux based distro. Have you tried installing it from AUR? Or at least check the PKGBUILD. There are listed all the dependencies you need. If you have `yay` installed you only need to issue `yay -S headsetcontrol-git` as normal user.

That worked but still curious why I got this error. I was able to compile before without any issues.

Sapd, 2021-01-02:
The error arises, as it simply cant find make in the path. You can try typing `where make` to see if it is in the PATH. If not, cmake cant find it either.

vitor180396, 2021-01-03:
> The error arises, as it simply cant find make in the path. You can try typing `where make` to see if it is in the PATH. If not, cmake cant find it either.

[image]

vscsilva, 2021-07-01:
You can close this.

parkerlreed, 2021-07-02:
What was it? (Since you are the author you can close this yourself as well)

vscsilva, 2021-07-02:
If I remember correctly that was due to missing `g++`

closed 2021-07-02 by vscsilva