Sapd/HeadsetControl #125: SteelSeries Arctis 7 - battery level issue

existme, 2021-01-02:
Hi,

I have `SteelSeries Arctis 7` and I have been using this small utility to query the battery level of the headset.
I have been experiencing several issues which I am listing here:

1. Sometimes the utility stays in a hang status and issuing `headsetcontrol -b` after displaying `Found SteelSeries Arctis (7/Pro)!` stays in an unresponsive mode. The only possible way to make the utility work again is to detach and reattach the USB dongle.
2. The battery level reported is sometimes completely off and the only way to make it report the correct battery level is to attach the charger while the headset is connected and let it stay for 10 seconds and then remove the charger again.

I looked at the source code but I can't see why this is happening and how to solve it. Any help would be appreciated.
Thanks

Sapd, 2021-01-02:
> Sometimes the utility stays in a hang status and issuing headsetcontrol -b after displaying Found SteelSeries Arctis (7/Pro)! stays in an unresponsive mode. The only possible way to make the utility work again is to detach and reattach the USB dongle.

After issuing -b, it will wait for the headset to write something. Maybe its a general implementation issue (e.g. missing initialisation code), or something specific on your OS part.

> The battery level reported is sometimes completely off and the only way to make it report the correct battery level is to attach the charger while the headset is connected and let it stay for 10 seconds and then remove the charger again.

That is actually quite normal. All the headsets simply seem to measure voltage (instead of summing up current over time, like on smartphones). As such the measurement is very inaccurate and can fluctuate. If the measurement is really far off, thats a sign that your battery will probably fail some time. When you plug in the charger for a short time, it will cause the voltage to rise quickly.

In most cases the accuracy is like only 4 steps I would say (e.g. full, 3/4, 2/4, > 1/4)

centic9, 2021-01-06:
The Arctis 7 works fairly well for me, I also fetch battery periodically, but it does never hang.

Battery levels mostly work, a few times it looked off, but it was mostly accurate enough to see when it was time to plug it in.

Maybe you can report exact OS versions, Kernel versions, udev-version ... to see if anything is fundamentally different in your case.

existme, 2021-01-07:
Hi,  thanks for your input. 

Here is the result of `uname -a`:

```
Linux  5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux.
```
How should I get the udev version? could you please instruct me on that.?

SebStaeubert, 2021-02-13:
I have a similar problem: the battery level stays at 38% and never changes regardless if the battery is fully charged or almost empty.
I tried HeadSetControl and even [trevarj/arctis7_controls](https://github.com/trevarj/arctis7_controls) - same results.
Any hints?
THX,Sebastian

jkms, 2021-05-26:
echoing this.  Series 7 battery is always off.  Windows reports correctly.

existme, 2021-06-25:
For solving the 1st issue I am currently using https://github.com/mcarans/resetusb/blob/master/reset_usb.py
I have put the python file in the path and I will issue:
``` sh
sudo reset_usb search Arctis
```

closed 2023-07-05 by github-actions[bot]