Sapd/HeadsetControl #128: Pro X Wireless "No supported headset found"

thoraxe, 2021-01-11:
I am trying to set the sidetone for the Pro X Wireless that I just got, but I am getting the "No supported headset found" error when running `headsetcontrol`.

Here's the info from `pactl list`:

```
Card #20
	Name: alsa_card.usb-Logitech_PRO_X_Wireless_Gaming_Headset-00
	Driver: module-alsa-card.c
	Owner Module: 40
	Properties:
		alsa.card = "4"
		alsa.card_name = "PRO X Wireless Gaming Headset"
		alsa.long_card_name = "Logitech PRO X Wireless Gaming Headset at usb-0000:00:14.0-3, full speed"
		alsa.driver_name = "snd_usb_audio"
		device.bus_path = "pci-0000:00:14.0-usb-0:3:1.0"
		sysfs.path = "/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/sound/card4"
		udev.id = "usb-Logitech_PRO_X_Wireless_Gaming_Headset-00"
		device.bus = "usb"
		device.vendor.id = "046d"
		device.vendor.name = "Logitech, Inc."
		device.product.id = "0aba"
		device.product.name = "PRO X Wireless Gaming Headset"
		device.serial = "Logitech_PRO_X_Wireless_Gaming_Headset"
		device.form_factor = "headset"
		device.string = "4"
		device.description = "PRO X Wireless Gaming Headset"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-headset-usb"
		device.intended_roles = "phone"
	Profiles:
		input:mono-fallback: Mono Input (sinks: 0, sources: 1, priority: 1, available: yes)
		input:multichannel-input: Multichannel Input (sinks: 0, sources: 1, priority: 1, available: yes)
		output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6500, available: yes)
		output:analog-stereo+input:mono-fallback: Analog Stereo Output + Mono Input (sinks: 1, sources: 1, priority: 6501, available: yes)
		output:analog-stereo+input:multichannel-input: Analog Stereo Output + Multichannel Input (sinks: 1, sources: 1, priority: 6501, available: yes)
		output:iec958-stereo: Digital Stereo (IEC958) Output (sinks: 1, sources: 0, priority: 5500, available: yes)
		output:iec958-stereo+input:mono-fallback: Digital Stereo (IEC958) Output + Mono Input (sinks: 1, sources: 1, priority: 5501, available: yes)
		output:iec958-stereo+input:multichannel-input: Digital Stereo (IEC958) Output + Multichannel Input (sinks: 1, sources: 1, priority: 5501, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: output:iec958-stereo+input:mono-fallback
	Ports:
		analog-input-mic: Microphone (type: Mic, priority: 8700, latency offset: 0 usec, availability unknown)
			Properties:
				device.icon_name = "audio-input-microphone"
			Part of profile(s): input:mono-fallback, output:analog-stereo+input:mono-fallback, output:iec958-stereo+input:mono-fallback
		analog-output: Analog Output (type: Analog, priority: 9900, latency offset: 0 usec, availability unknown)
			Part of profile(s): output:analog-stereo, output:analog-stereo+input:mono-fallback, output:analog-stereo+input:multichannel-input
		iec958-stereo-output: Digital Output (S/PDIF) (type: SPDIF, priority: 0, latency offset: 0 usec, availability unknown)
			Part of profile(s): output:iec958-stereo, output:iec958-stereo+input:mono-fallback, output:iec958-stereo+input:multichannel-input
```

I'm assuming it's because the product ID `0aba` doesn't match anything in the devices files.

This line mentions the ProX:
https://github.com/Sapd/HeadsetControl/blob/master/src/devices/logitech_gpro.c#L9

But it has a different ID (`0aaa`).

I'll try messing around with the source and recompiling for giggles to see if I can make it work and will report back.

thoraxe, 2021-01-11:
I created a `..._gprox` H/C file and added it to `src/devices/CMakeLists.txt`. I used the following address:

`static const uint16_t PRODUCT_ID = 0x0aba;`

I then did the build, and it's building that object:

```
...
[build] [ 86%] Building C object CMakeFiles/headsetcontrol.dir/src/devices/logitech_gprox.c.
...
```

However, I am still getting `No supported headset found`.

Sapd, 2021-01-11:
You also have to call the init function of your new file / headset in src/device_registry.c

thoraxe, 2021-01-11:
OK, I've made progress, but I think now I'll likely need to dig into the USB stuff, as my guess is this device doesn't behave the same as the Pro:

```
Found Logitech G PRO X!

Couldn't open device.
```

thoraxe, 2021-01-11:
never mind - i have to run it with sudo! PR incoming! sidetone works!

thoraxe, 2021-01-11:
Fixed by #129

closed 2021-11-13 by Sapd