Sapd/HeadsetControl #131: Pro X / Pro battery support

thoraxe, 2021-01-12:
This is an issue to discuss figuring out battery support on the Pro / Pro X wireless headsets.

The G933-esque battery support came from https://github.com/ashkitten/g933-utils

That repo includes a Wireshark Lua plugin to help analyze the Logitech HID+ protocol, but it seems to have a few shortcomings with the newer G Hub software and the Pro X.

https://github.com/ashkitten/g933-utils/issues/17

I didn't see anywhere in the "known" messages where the battery status was queried by the G-Hub software. I tried the following:

* start wireshark with dongle connected but headset power off
* open g-hub application
* turn on headset

At this point, the headset shows up as connected in G-Hub and the battery level appears. So, somehow the app is determining the battery percentage somehow. 

The G933 implementation has a get voltage function that it then converts to a percentage:
https://github.com/Sapd/HeadsetControl/blob/master/src/devices/logitech_g633_g933_935.c#L51-L89

However, as I mentioned earlier, the Pro X doesn't appear to ever be sent any kind of get battery request that's "known" at this point.

In the issue I filed with ashkitten's repo, you can see the Lua issues I'm having.

If you have any suggestions or ideas, let me know. I have the device so it's easy for me to poke at it.

Sapd, 2021-01-12:
I'd simply try it without that plugin. Wireshark is quite powerful and has a lot of filters.

thoraxe, 2021-01-13:
Right, but the plugin is what is dissecting the packet capture and explaining what is being sent to/from the headset.

Here's a capture of turning the headset on and waiting and then turning it off.
https://drive.google.com/file/d/11r9voCayKynHl0V2QVMm9RHaIVcexw_p/view?usp=sharing

markpash, 2021-08-13:
@thoraxe I just downloaded your pcap and went through it as I don't feel like booting into windows. I think I may have found the message that requests the battery info.

Luckily the response is identical to the G933/G935 response, so after a few attempts, I think I found the correct request and tested it with my headset. It's able to correctly identify if it's charging or not. I'm not sure if the voltage calculation is correct since I just copied it from the G933 code. I can make a PR for this, but this may need to be a separate device, as the normal wired headset may not support getting battery info.

@Sapd would you be willing to review a PR if I make one?

Sapd, 2021-11-13:
> @Sapd would you be willing to review a PR if I make one?

Sure, feel free to open up a PR any time. Things can then still be completely changed or improved.

closed 2023-07-03 by github-actions[bot]