Sapd/HeadsetControl #133: Arctis 7 lights toggle not working

jzbor, 2021-01-15:
I am currently using the Arctis 7 with Manjaro. The lights toggle (`headsetcontrol -l 0/1`) does not seem to have any effect on neither the headset itself nor the wireless dongle. Strangely though headsetcontrol reports `Success!`. I have the udev stuff running with udev (systemd) version 247.2.

lortimer, 2021-03-13:
@jzbor Does it turn the wireless dongle's light off when the headset is off? I have just installed this utility, and I found that the light is on when the headset is connected, but is off when the headset is off. Usually, it would flash.

lortimer, 2021-03-13:
@Sapd How did you figure out what bytes to write to the Arctis7 to toggle the light? Do you know if it would be possible to turn it off whether or not the headset is on?

I see the switch_lights function, but I can't find any Arctis documentation about the bytes in the report. I know the first byte is the [length of the report](http://hidapi-d.dpldocs.info/hidapi.bindings.hid_write.html), and it' implied that 0x00 at the end means "off" or 0x02 means "on".

```c++
static int arctis_7_switch_lights(hid_device* device_handle, uint8_t on)
{
    unsigned char data[8] = { 0x06, 0x55, 0x01, on ? 0x02 : 0x00 }; // length 6, ?, ?, on/off
    int ret = hid_write(device_handle, data, 8);

    if (ret >= 0) {
        ret = arctis_7_save_state(device_handle);
    }

    return ret;
}
```

Thanks for this awesome utility! Installation was easy and it did what I wanted!

centic9, 2021-03-13:
This LED-switching functionality for the Arctis 7 was contributed by me. 

The protocol is not documented at all and the Steelseries refused to provide any specification when I contacted them. So it was all found out via reverse engineering by multiple developers of the various opensource tools that are available. I have some notes if you are interested in the details that I collected over time.

The effect of the LED switch for the Arctis 7 only takes effect when the headset itself is actually turned off:
* When LED=1, the LED on the wireless dongle is blinking on and off
* When LED=0, the LED on the wrieless dongle is off
* When the Headset is turned on, the LED on the dongle is always on.

So this switch allows to avoid the blinking of the LED on the dongle, which is an annoying distraction for me whenever I have the headset turned off.

I didn't see any other LED on the device and I also did not find any other function than this switching of blinking or not blinking.

lortimer, 2021-03-14:
Thank you @centic9! Impressive work, kudos to all of you for reverse-engineering it! I don't think you adding your notes here will solve @jzbor's problem, unless I didn't understand it. I think this information makes it clear what can be done with the LED.

jzbor, 2021-03-16:
Thanks for the great work @centic9! I am sorry I had totally forgotten about this issue. I dont know why but since about a month it works now. I was pretty sure I double checked and restarted everything back then, but maybe just needed another restart. Sorry for bothering. Thanks for bringing this amazing support for the Arctis 7 to Linux!

jzbor, 2021-03-16:
@lortimer As I understood turning of the flashing is exactly what it should do - somehow it didnt work back then... if you are interested in always disabling the LED I guess it would deserve its own issue. Feel free to reopen this one though if you disagree.

closed 2021-03-16 by jzbor