Sapd/HeadsetControl #137: Corsair Void RGB Elite cannot set sidetone

dwalton76, 2021-02-19:
I am running ubuntu 20.04 with a [Corsair Void RGB Elite](https://www.amazon.com/gp/product/B07XCYT7HV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) but I cannot set the sidetone
```
$ headsetcontrol -s 16
Found Corsair Headset Device!

ioctl (SFEATURE): Broken pipe
Failed to set sidetone. Error: -1: (null)
$ 
```

`lsusb -v` shows
```
Bus 001 Device 066: ID 1b1c:0a52 Corsair Corsair VOID ELITE USB Gaming Headset
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0a52 
  bcdDevice           20.41
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x013e
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
[... 270 lines cut ...]
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     308
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
```

`headsetcontrol -n 0` and `headsetcontrol -n 1` both work...so do `headsetcontrol -l 0` and `headsetcontrol -l 1`.

Any ideas?

nkenschaft, 2021-03-15:
I'm having the same issue, try looking at issue #69

Simon1511, 2021-05-18:
Same thing here with Void Pro USB on Ubuntu 20.04.

Sapd, 2021-11-13:
Does the issue still exist? Maybe a different interfaceID is needed.

Simon1511, 2021-11-13:
> Does the issue still exist? Maybe a different interfaceID is needed.

Yes, the error is still the same with the latest version.

dwalton76, 2021-11-15:
I no longer have the laptop where I saw this so I will defer to @Simon1511

Tumpes, 2022-04-26:
Any updates? I have the problem on ubuntu 20.04 using a void pro rgb wired.

Jerther, 2023-02-14:
Still happening :(
Fedora 36

`-l` and `-n` work fine though.

I can set the sidetone with alsamixer, as described in other issues. The slider is just misnamed:

[image]

Sapd, 2023-05-01:
Closing, the sidetone is controlled via an audio channel, which we can't access via HID.

closed 2023-05-01 by Sapd