Sapd/HeadsetControl #145: Corsair Virtuoso SE compatibility

jonathanlaf, 2021-03-30:
Hello, 

I tried to configure my CORSAIR VIRTUOSO SE to see if it's compatible, and I was wondering if I done it right. If yes, it's not compatible, and would like to know if it's possible to make it compatible. How can I help? 

`lsusb`
`Bus 001 Device 014: ID 1b1c:0a40 Corsair CORSAIR VIRTUOSO SE Wireless Gaming Headset`

`HeadsetControl/src/devices/corsair_void.c`
`#define ID_CORSAIR_VIRTUOSO_SE_WIRELESS 0x0a40`

```
cd ../../build
make
make install
make install
headsetcontrol -s 0
```
`No supported headset found`

themoosman, 2021-03-30:
Look here:  https://github.com/Sapd/HeadsetControl/issues/71

jonathanlaf, 2021-03-30:
> Look here: #71

Thank you! I'll participate in this thread, I have a Windows computer and an Ubuntu computer, so I might be helpful. I'm also a programmer.

closed 2021-03-30 by jonathanlaf