Sapd/HeadsetControl #150: Request support for Epos GSP 370 headset

nt74, 2021-05-07:
I wonder if it is possible to add support for the wireless headset from EPOS, model GSP 370?
In the Epos Gaming Suite software for Windows there is possibilities to see current battery level, headset/dongle firmware, audio equalizer,  surround effect mode 2.0 / 7.1, voice enhancer (off/warm(clear).

The most important feature would be to see the current battery status and audio mode (2.0 / 7.1).

nt74, 2021-05-11:
Please let me know if you want me to capture usb messages with wireshark, or assist in any possible way?

Sapd, 2021-05-12:
Here is a tutorial for using Wireshark: https://github.com/Sapd/HeadsetControl/wiki/Development
You can try that and report your findings

nt74, 2021-05-18:
I have captured with Wireshark the USB dongle device uses bus 2 device 7.
How can I read out variables from this capture?

[gsp370_capture1.zip](https://github.com/Sapd/HeadsetControl/files/6501284/gsp370_capture1.zip)

Sapd, 2021-05-18:
In the link I sent you, there is a tutorial of how to get e.g. the sidetone.

You also need to specify what you did in your recording.

nt74, 2021-05-18:
Here are two files with the same bus (2) and device (7), testing sidetone up/down and vice versa a few times then leaving it to 100%. The second file is testing to turn volume up/down and vice versa on the headset volume controller.
[gsp370_sidetone_testing.zip](https://github.com/Sapd/HeadsetControl/files/6502368/gsp370_sidetone_testing.zip)
[gsp370_volume_sliding.zip](https://github.com/Sapd/HeadsetControl/files/6502369/gsp370_volume_sliding.zip)

Sapd, 2021-05-18:
I see a lot of Audio data. Make sure to not play any audio or hear music while capturing.

When starting capturing, it should look something like this (choose only your device, make sure to use USBPCAP):
[image]

nt74, 2021-05-18:
Here are the same things but with no audio.
[gsp370_sidetone_noaudio.zip](https://github.com/Sapd/HeadsetControl/files/6503384/gsp370_sidetone_noaudio.zip)
[gsp370_volume_noaudio.zip](https://github.com/Sapd/HeadsetControl/files/6503385/gsp370_volume_noaudio.zip)

nt74, 2021-05-27:
Here is a captured sidetone going from mic input vol 0 to 100. The usb device is 1.8.x (captured on a laptop)
[gsp370_sidetone_min-to-max_dev1.8.x.pcapng.zip](https://github.com/Sapd/HeadsetControl/files/6552758/gsp370_sidetone_min-to-max_dev1.8.x.pcapng.zip)

closed 2023-07-02 by github-actions[bot]