Sapd/HeadsetControl #152: Logitech G733 Gaming Headset - not recognized

hselasky, 2021-05-08:
usbconfig -d ugen0.5 dump_device_desc
ugen0.5: <Logitech G733 Gaming Headset> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)

  bLength = 0x0012 
  bDescriptorType = 0x0001 
  bcdUSB = 0x0110 
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000 
  bDeviceProtocol = 0x0000 
  bMaxPacketSize0 = 0x0040 
  idVendor = 0x046d 
  idProduct = 0x0ab5 
  bcdDevice = 0x0100 
  iManufacturer = 0x0003  <Logitech>
  iProduct = 0x0004  <G733 Gaming Headset>
  iSerialNumber = 0x0000  <no string>
  bNumConfigurations = 0x0001

DanyGee, 2021-06-01:
Did you try running `headsetcontrol` as root?

hselasky, 2021-06-01:
Yes, the USB ID's are missing in there.

Sapd, 2021-11-13:
The most recent version should have the ID included

closed 2021-11-13 by Sapd

hselasky, 2021-11-13:
Thank you! Works as expected!