Sapd/HeadsetControl #154: Version 2.4 missing DLL on Windows

guaycuru, 2021-05-13:
I'm receiving this error when trying to run version 2.4 on Windows 10 Pro x64:
`The code execution cannot proceed because libwinpthread-1.dll was not found. Reinistalling the program may fix this problem.`

Sapd, 2021-05-13:
Try this one. It includes all dependencies statically. Should also work without the hidapi.dll 

If it works for you I make a new release 

[headsetcontrol.exe.zip](https://github.com/Sapd/HeadsetControl/files/6472627/headsetcontrol.exe.zip)

guaycuru, 2021-05-13:
It "almost" worked:

```
> headsetcontrol.exe -b
Found Corsair Headset Device!

Failed to request battery. Error: -1: Incorrect function.
```

Sapd, 2021-05-13:
Did it work before with the old version?

And do other things work, like sidetone or `-n 0`?

guaycuru, 2021-05-13:
Version 2.3 did not find my headset.
All commands seem to have the same issue:
```
>headsetcontrol.exe -n 0
Found Corsair Headset Device!

Failed to send notification sound. Error: -1: Incorrect function.
```

Sapd, 2021-05-13:
Thats difficult then. It means that the usagepage and usageid for your headset differs (and Windows cares for that).

You basically have to download the testgui from HIDAPI https://github.com/signal11/hidapi/releases 
it is located in the folder `hidapi-0.7.0.zip\hidapi-0.7.0\testgui` it should show all usagepage and usageids.

guaycuru, 2021-05-13:
Ok I tried that but I'm not sure where to find "all usagepage and usageids"...
This is what I have:
[image]

Sapd, 2021-05-13:
> "all usagepage and usageids".

They are after the word *usage* inside your screenshot. So you have a pair of three. The last one (ff00:0001) is the one in the current implementation

Try copy pasting this besides Send *Feature* Report:
`0xff 0xb 0 0xff 0x4 0xe 0xff 0x5 0x1 0x4 0 0xfa 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0`

Then click on the button, and do this for all 3 entrys, and see if it works for one of them.

guaycuru, 2021-05-13:
It seemed to work (at least no error was shown) for the last device: `ff00:0001`

Sapd, 2021-05-13:
Ah I think I know then what the issue is.

Did you explicitly try Sidetone?
e.g.
`headsetcontrol -s 128`

guaycuru, 2021-05-13:
Sidetone works!

```
>headsetcontrol.exe -s 128
Found Corsair Headset Device!

Success!
```

Sapd, 2021-05-13:
I see.

The problem is, the different features are basically in different so called usage-pages (the different ids you see). That means the software can currently only support one feature at a time, as it wasn't developed with that in mind (as on Linux, it does not matter which usage-page you use, as there is none). So unfortunately you can only use one feature without recompiling the software.

guaycuru, 2021-05-13:
Oh I see. That's a bummer but I understand. Thank you!

guaycuru, 2021-05-24:
How would I go about changing the usage-page to support for instance turning the LED on/off? I know I need to recompile after, so my question is what portion of the code I need to change to accomplish that?

Sapd, 2021-05-24:
You need to change those lines in corsair_void.c

https://github.com/Sapd/HeadsetControl/blob/9011bc5fbb514377059d3505204f2284aa46794a/src/devices/corsair_void.c#L55-L56

vcekron, 2021-11-10:
> The new release has everything statically linked on windows and can be run directly: https://github.com/Sapd/HeadsetControl/releases

Sadly, this behaviour remains for me.
* Headset: Corsair Void RGB
* Release 2.5
* Windows 11

Example:
```
headsetcontrol.exe -b
Found Corsair Headset Device!

Failed to request battery. Error: -1: Incorrect function.
```
```
headsetcontrol.exe -s 0
Found Corsair Headset Device!

Success!
```

Sapd, 2021-11-10:
> > The new release has everything statically linked on windows and can be run directly: https://github.com/Sapd/HeadsetControl/releases
> 
> Sadly, this behaviour remains for me.

This was not addressed by this release. I already have an idea of how to solve it, however it will still take a bit of time

vcekron, 2021-11-10:
> > > The new release has everything statically linked on windows and can be run directly: https://github.com/Sapd/HeadsetControl/releases
> > 
> > 
> > Sadly, this behaviour remains for me.
> 
> This was not addressed by this release. I already have an idea of how to solve it, however it will still take a bit of time

Ah, my bad! Thanks for the clarification :)

Sapd, 2021-11-13:
> Ah, my bad! Thanks for the clarification :)

@vcekron The new version completely fixed it. Should now work out of the box with all features. Feel free to create a new issue if there are still problems