Sapd/HeadsetControl #157: G733 is always showing 100% battery while discharging.

DanyGee, 2021-06-01:
G733 is always showing 100% battery while discharging. 
(even after unpluging the cable, restarting the headset, restarting the system)
```
sudo headsetcontrol -b
Found Logitech G633/G635/G933/G935!

Battery: 100%
```
Plugin the USB cable in results the state change to proper:
```
sudo headsetcontrol -b
Found Logitech G633/G635/G933/G935!

Battery: Charging
```
Tested on system:

```
OS: Zorin OS 16 x86_64 
Kernel: 5.8.0-53-generic 
Packages: 2209 (dpkg), 26 (flatpak), 6 (snap)
Shell: bash 5.0.17 
Resolution: 3840x2160 
DE: GNOME 
WM: Mutter 
WM Theme: ZorinBlue-Light 
Theme: ZorinBlue-Light [GTK2/3] 
Icons: Papirus-Light [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i7-6700K (8) @ 4.300GHz 
GPU: NVIDIA GeForce GTX 1080 
Memory: 4018MiB / 64257MiB 
```
If any additional info is needed - let me know.

Qurtison, 2022-09-25:
Have the same issue, is there anything I can do to help debug this?

lezcano17, 2022-10-23:
I have a similar issue, mine always shows a percentage that is much higher than the real one. Around 20-30% off.

neoxpert, 2023-08-22:
Mine is always showing more than it should. When it shows 95% it is actually at 90%. When it shows 50% it's actually almost 0% already.

closed 2024-08-17 by github-actions[bot]