Sapd/HeadsetControl #158: Corair HS70 BLUETOOTH

Kharonus, 2021-06-01:
Ahoi,
already looked to some of the other issue. Apparently HS70 works fine for most users, after maybe having some starting problems.

my system: ubuntu 20.04
my headset (from lsusb): `Bus 001 Device 006: ID 1b1c:0a61 Corsair HS70 BLUETOOTH Gaming Headset`

What goes wrong on my side:
```
> headsetcontrol -b
Found Corsair Headset Device!

Failed to request battery. Error: -1: (null)
```

same goes for setting sidetone, playing notification, etc.

I added the device id like written in corsair HOWTO to the corsair_void.c, run `make` and `make install`.
In addition, my microphone seems to fail to work at all, I cannot record any input.

Do you by chance have a clue, what I'm missing?

Sapd, 2021-11-13:
Probably same protocol/problem like: https://github.com/Sapd/HeadsetControl/issues/178

bnjns, 2021-11-13:
I've got both the HS70 and HS80 so can give the 70 a try too

closed 2023-07-02 by github-actions[bot]