Sapd/HeadsetControl #159: Missing Boost license file

mohanboddu, 2021-06-08:
cmake_modules/Findhidapi.cmake is licensed under [Boost Software License 1.0](https://www.boost.org/LICENSE_1_0.txt), as per  the license the license file `must be included in all copies of the Software`.

Please add the missing license file to the repo.

Sapd, 2021-11-13:
The boost license file is linked, which should be according to the law here enough and - as it is solely hosted online - can be interpreted as included.

closed 2021-11-13 by Sapd

carlwgeorge, 2021-11-13:
> The copyright notices in the Software and this entire statement, including
the above license grant, this restriction and the following disclaimer,
**must be included in all copies of the Software**, in whole or in part, and
all derivative works of the Software, unless such copies or derivative
works are solely in the form of machine-executable object code generated by
a source language processor.

A link to the license is not the same thing as including the license.  Please comply with the license.  My suggestions:

1. In your existing license file, include the text of both the Boost license and the GPLv3, with comments explaining which license applies to which files.
2. Rename the existing license file to LICENSE-GPLv3, and copy the Boost license text into a file named LICENSE-Boost.  Write a new file named LICENSE that explains which license applies to which files.

Sapd, 2021-11-14:
I included the license explicitly into the offending source file.

Should the author of FindHidapi.cmake have suggestions to do this differently, he can contact me at any time (either via Mail or LinkedIn).