Sapd/HeadsetControl #164: Arctis 7p not recogniced

auduny, 2021-06-17:
Steelseries Arctis 7p is not recognised running on os x 10.15.7

Anything I can do to help?

Sapd, 2021-11-13:
You could try adding the productID of your device to `steelseries_arctic_7.c`

MattDiesel, 2022-01-07:
The device ID for mine is 0x220e. I added it and rebuilt, and it picks up the device but the commands fail with Error: -1: Broken pipe (this is on linux). I'll have a play and see if I can figure it out.

ikogan, 2022-02-01:
So as I discovered after getting this, the Arctis 7 Plus is different than the Arctis 7 or Pro. We have the same device ID. It looks like this'll take some work.

ribugent, 2022-02-02:
Same here but with the id `0x12d5`
```
Device Found
  VendorID: 0x1038
 ProductID: 0x12d5
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7P
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1
```

ikogan, 2022-02-04:
So the code seems to be using `hid_write` to speak to the supported Arctis headsets but when I sniff the USB for this one, I see that it's using `SET_REPORT`. The docs say to use `hid_send_feature_report` for those. I'm not sure if this is a dramatic difference requiring a totally different device rather than just adding the IDs to the existing implementation. Anyway, here's some info from the USB sniff.

### Sidetone

Off: 
```
0000   1c 00 20 ca 9c 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 39 00 00 00 00 00 00 00 00 00 00 00
```
Low:
```
0000   1c 00 10 f0 6d 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 39 01 00 00 00 00 00 00 00 00 00 00
```
Medium:
```
0000   1c 00 60 c2 35 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 39 02 00 00 00 00 00 00 00 00 00 00
```
High:
```
0000   1c 00 20 0b e8 7e 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 39 03 00 00 00 00 00 00 00 00 00 00
```

### Inactive Length

Never:
```
0000   1c 00 20 db 01 81 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 00 00 00 00 00 00 00 00 00 00 00
```
1 Minute:
```
0000   1c 00 b0 29 fc 81 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 01 00 00 00 00 00 00 00 00 00 00
```
5 Minutes:
```
0000   1c 00 80 b8 39 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 05 00 00 00 00 00 00 00 00 00 00
```
10 Minutes:
```
0000   1c 00 e0 09 89 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 0a 00 00 00 00 00 00 00 00 00 00
```
15 Minutes:
```
0000   1c 00 80 b8 39 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 0f 00 00 00 00 00 00 00 00 00 00
```
30 Minutes:
```
0000   1c 00 70 ba 11 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 1e 00 00 00 00 00 00 00 00 00 00
```
45 Minutes:
```
0000   1c 00 20 e7 9c 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 2d 00 00 00 00 00 00 00 00 00 00
```
60 Minutes:
```
0000   1c 00 50 e2 c7 7e 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 3c 00 00 00 00 00 00 00 00 00 00
```
75 Minutes:
```
0000   1c 00 70 ba 11 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 4b 00 00 00 00 00 00 00 00 00 00
```
90 Minutes:
```
0000   1c 00 10 50 00 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 a3 5a 00 00 00 00 00 00 00 00 00 00
```
### Mic Volume

Lowest Value:
```
0000   1c 00 10 f0 38 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 37 01 00 00 00 00 00 00 00 00 00 00
```
Middle Value:
```
0000   1c 00 10 a0 e8 81 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 37 04 00 00 00 00 00 00 00 00 00 00
```
Highest Value
```
0000   1c 00 10 d0 6e 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 37 07 00 00 00 00 00 00 00 00 00 00
```
### LED

There do not seem to be LED settings for this device.

### ChatMix

ChatMix seems to be controlled by a physical dial and not by any software options that I can see.

### Equalizer

There does seem to be an equalizer that sends USB reports when toggled

### Battery

I'm not sure how to determine where the battery read is. A few reports are sent when it's turned on:

```
0000   1c 00 00 09 05 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```
```
0000   1c 00 00 09 05 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```
```
0000   1c 00 a0 04 5d 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```
```
0000   1c 00 b0 d9 37 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```
```
0000   1c 00 10 10 7f 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```
```
0000   1c 00 00 09 05 82 86 b0 ff ff 00 00 00 00 1b 00
0010   00 03 00 01 00 00 02 48 00 00 00 00 21 09 00 02
0020   03 00 40 00 10 00 00 00 00 00 00 00 00 00 00 00
```

Finally, it seems that the device `220e` is the wireless dongle and `220c` is the headset itself when connected over USB. The headset only connects to the app when the dongle is plugged in.

Sapd, 2022-02-04:
> So the code seems to be using `hid_write` to speak to the supported Arctis headsets but when I sniff the USB for this one, I see that it's using `SET_REPORT`. The docs say to use [...]

Indeed it needs a new device file. Probably all packets are also different.

Michsior14, 2022-05-04:
For all that are interested in 7 plus, the support for this headset has landed on master via #209.

ikogan, 2022-05-05:
Can confirm this is working on master! It looks like sidetone, battery status, and inactive time are all supported. Equalizer support would be nice but I think this issue can be closed.

Michsior14, 2022-05-05:
@ikogan For equalizer support you can try #210.

closed 2023-05-01 by Sapd